odelot/aws-mqtt-websockets #2: MQTT fails on json/quote?

svdgraaf, 2016-03-25:
I think I've found an issue. When I publish a message with a quote it in `"`, the message is never received on the iot end. When I publish a message without json, it does work:

The topic I subscribed to is `$aws/things/xyz/shadow/update`.

With regular text:
```
connecting to wifi
Connected
socket connected
MQTT connecting
MQTT connected
MQTT subscribed
Message 1 arrived: qos 1, retained 0, dup 0, packetid 2
Payload 
Message 2 arrived: qos 1, retained 0, dup 0, packetid 4
Payload foobar
Message 3 arrived: qos 1, retained 0, dup 0, packetid 5
Payload {}
Message 4 arrived: qos 1, retained 0, dup 0, packetid 6
Payload foo
```

With quotes:
```
connecting to wifi
Connected
socket connected
MQTT connecting
MQTT connected
MQTT subscribed
```

No response ;)

svdgraaf, 2016-03-25:
Hmmm, after some more testing, it seems to be that is something weird going on with the length of the string. If publish `{"barrr"}`, it doesn't seem to be receiving anything, but after a while, the esp will reboot (probably buffer overflow?). If I publish a slightly shorter string `{"barr"}`, it just works.

I'll see if I can find why.

svdgraaf, 2016-03-25:
I can confirm that it works both ways (receiving and publishing), longer messages are not processed...

svdgraaf, 2016-03-25:
Uuuuuugh, got it! Took me a while! :D

Topic needs to be a `const char*`, it's probably used a lot, so not passing it by reference will eat up quite some memory probably ;)

Awesome, it works! :D woohoo!

svdgraaf, 2016-03-28:
Hmmm, there still is an issue with either long messages, or a long `topic` name. If a have a long Topic name, eg: `$aws/things/your-device/shadow/update`, and a long message: `{"reported":{"foo": "bar"}, desired:{}}`. Nothing will be delivered.
If I have a short message and a long topic, it works. A short topic and a long message wil work too. I haven't found where the bug is yet, or what the maximum amount of characters is.

svdgraaf, 2016-03-28:
Ha! Got it :)

It was a bit of a search, but the Paho client is by default limited to max packages of X size. So ofcourse, this would result message failures when the topic + message gets a bit longer.

Changing:
`MQTT::Client<IPStack, Countdown, 50, 1> client = MQTT::Client<IPStack, Countdown, 50, 1>(ipstack);`
into:
`MQTT::Client<IPStack, Countdown, 255, 1> client = MQTT::Client<IPStack, Countdown, 255, 1>(ipstack);`

Fixed the issue, the 255 is the max length of the MQTT packages (the longer, the memory it will use as well, ofcourse). This is also valid for the incoming package length (so for receiving messages from the outside). Pretty important :)

closed 2016-03-28 by svdgraaf

odelot, 2016-03-28:
@svdgraaf I've notice this testing today. I've implemented the auto-reconnection feature and fix the need of that wait for connection workaround. Of course, it would be very nice to stress test these features ;-)

next target.... remove the need for bytebuffer

svdgraaf, 2016-03-28:
Awesome, I'll run some more testing tomorrow :)

Cheers!
Sander

odelot, 2016-04-01:
hi @svdgraaf, let me ask you something?

are you using the stage or stable version of arduino esp8266 code? 2.1.0 or 2.1.0-rc2?

I am investigating the memory leak after many reconnections. Tried the stable version and nothing worked, neither the first connection. O.o

svdgraaf, 2016-04-02:
I'm running a git checkout, it's revision `633e48f3aec5f1c3c11d4498fc90d378d49e6e9f`. I haven't had these issues yet (on a Adafruit huzzah board).

odelot, 2016-04-03:
I've got the latest version from git and I'm still having problems

this simple code shows that heapspace get smaller after each reconnection

`       
        const char* host = "google.com";
	WiFiClient c;
	for (int i =0; i< 100; i+=1) {
		Serial.print (i);
		Serial.print ("-");
		Serial.println (ESP.getFreeHeap());
		if (c.connect (host,80)) {
			delay (100);
			c.flush ();
			c.stop ();
			delay (100);
		} else {
			Serial.println ("connection refused");
		}
	} `

this code is using esp8266 core :-(  my board is a generic one but It seems to be a software issue.

daniele-salvagni, 2016-04-03:
I tested that code and I had the same outcome, the heap keeps shrinking after every loop. I did a quick search and it looks like that you are not the first one to notice this, here is a blog post about that (see *"The Unexpected Discovery"* paragraph): http://internetofhomethings.com/homethings/?p=1232

And couple issues I found showing more details and possible solutions: https://github.com/esp8266/Arduino/issues/230, https://github.com/esp8266/Arduino/issues/1070

odelot, 2016-04-04:
thank you @daniele-salvagni, this mod fixed the issue https://github.com/esp8266/Arduino/issues/230#issuecomment-116816950

strange that it is not in the master branch yet.

svdgraaf, 2016-04-08:
I can concur that I'm having the same issue, I just did a recompile with the `tcp_abort(_pcb);` fix, let's see how long it will run.

btw: I had issues with connecting to the mqtt from the latest checkout on the `master` esp8266 branch, I'm back to my above mentioned hash now, and everything works like a charm so far.

odelot, 2016-04-08:
I'm still investigating before close issue https://github.com/odelot/aws-mqtt-websockets/issues/5

`tcp_abort(_pcb) ` improved, but I am still losing memory. I am looking at the moment in the `getCurrentTime ` function. Maybe it is a good idea to reuse the `WifiClient `object. It seems that we lose some memory there. The problem is find free time to work on this :-( I will keep you informed about my findings

svdgraaf, 2016-04-12:
I'm now posting the heap size in my latest build to AWS as well, and plotting it into Kibana: [image]

The line is pretty lineair, so that's good. I will see if I can find the time to debug the `getCurrentTime()` method. I saw some other library which used a ntp server for that. Might be a bit overkill though.

odelot, 2016-04-12:
@svdgraaf it as a great way to keep track of memory

Here is my findings:

Ive cleaned up the code a little bit (commited it already*) and changed the example (see below).

Now it seems that memory usage is almost ok, althought we need to test and investigate a little more. `getCurrentTime` is ok as well

Between the first and second connection, we lose 316 bytes (maybe the first subscribe? need to investigate it). After this, the memory seems to float but coming back eventually to the same level that was recorded in the second connection. When we got an error trying to connect to mqtt server, we lose 168 bytes as well.

```
connected
51191 - conn: 1 - (27744)
websocket layer connected / MQTT connecting / MQTT connected / MQTT subscribed
362290 - conn: 2 - (27408)
websocket layer connected / MQTT connecting / MQTT connected / MQTT subscribed
672864 - conn: 3 - (27408)
websocket layer connected / MQTT connecting / MQTT connected / MQTT subscribed
984979 - conn: 4 - (27408)
websocket layer connected / MQTT connecting / MQTT connected / MQTT subscribed
1298342 - conn: 5 - (13176)
websocket layer connected / MQTT connecting / MQTT connected / MQTT subscribed
1611359 - conn: 6 - (13176)
websocket layer connected / MQTT connecting / MQTT connected / MQTT subscribed
1922638 - conn: 7 - (27408)
websocket layer connected / MQTT connecting / MQTT connected / MQTT subscribed
2233730 - conn: 8 - (27408)
websocket layer connected / MQTT connecting / 
error connection to MQTT server2
2241845 - conn: 9 - (13176)
websocket layer connected / MQTT connecting / MQTT connected / MQTT subscribed
2551782 - conn: 10 - (27240)
websocket layer connected / MQTT connecting / MQTT connected / MQTT subscribed
2862456 - conn: 11 - (27240)
websocket layer connected / MQTT connecting / MQTT connected / MQTT subscribed
```

and here is my new example code (still didnt commit it*, I need to test more):

- note that I've used 6000 bytes as circular buffer between WS layer and MQTT layer and 2056 bytes as max mqtt message lenght... these are far more than necessary, but help me to accelerate the soft reset caused by the memory leak
-  I transformed the mqtt client as a pointer to test its destroyer. I noticed that when mqtt connection got an error about 10000 bytes was lost, so I hope destroying it every connection helps us to avoid this. (interesting that after switching to pointer, we still lost 168 bytes when MQTT server could not connect)

\* I probably should make a branch to avoid been commiting all the time. Sorry, I'am used to CVS, SVN, but git is still new to me. I will try to learn all git stuff in the near future ;-)

```
#include <Arduino.h>
#include <Stream.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

//AWS
#include "sha256.h"
#include "Utils.h"
#include "AWSClient2.h"

//WEBSockets
#include <Hash.h>
#include <WebSocketsClient.h>

//MQTT PAHO
#include <SPI.h>
#include <IPStack.h>
#include <Countdown.h>
#include <MQTTClient.h>

//AWS MQTT Websocket
#include "Client.h"
#include "AWSWebSocketClient.h"
#include "CircularByteBuffer.h"
[... 157 lines cut ...]
    awsWSclient.setAWSDomain(aws_endpoint);
    awsWSclient.setAWSKeyID(aws_key);
    awsWSclient.setAWSSecretKey(aws_secret);
    awsWSclient.setUseSSL(true);

    if (connect ()){
      subscribe ();
      //sendmessage ();
    }

}

void loop() {
  //keep the mqtt up and running
  if (awsWSclient.connected ()) {    
      client->yield();
  } else {
    //handle reconnection
    if (connect ()){
      subscribe ();      
    }
  }
    
}
```

odelot, 2016-04-14:
after 28 hours, the esp8266 crashed again... but it seems not be caused by memory leak

I will try to investigate this exception (29) but the memory issue seems to be under control

have any of you guys had already problems with exception 29?
```
100187627 - conn: 372 - (27240)
websocket layer connected / MQTT connecting / MQTT connected / MQTT subscribed
100498617 - conn: 373 - (27240)
websocket layer connected / MQTT connecting / MQTT connected / MQTT subscribed
100809295 - conn: 374 - (27240)
websocket layer connected / MQTT connecting / MQTT connected / MQTT subscribed

Exception (29):
epc1=0x4000e1b2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

```
*note that memory usage (27240) from connection 374 is the same from the connection from last post

svdgraaf, 2016-04-14:
Nice! I haven't been able to look at your new code yet, but sounds very promising! :)

Groeten!
Sander

svdgraaf, 2016-04-16:
After some looking around, the 29 exception does point to an invalid reference somewhere (trying to overwrite a value in virtual memory which is prohibited). Can you perhaps create push a branch with your current code? I can hook up a debugger, and see where the issue is.

svdgraaf, 2016-04-16:
I figured my esp must have crashed by now, but hadn't checked the actual output. I just did, and it's still running (~29 hours now), the heap size is stable for the last few hours around 10k. It's posting to a mqtt channel every 10 seconds. I'm running `633e48f3aec5f1c3c11d4498fc90d378d49e6e9f` of the esp8266 repo, with the manual 'fix' for the `txp_abort()`. Let's see how long it will last...

[image]

svdgraaf, 2016-04-17:
Still going strong after 49 hours... Heap size still the same

profFabio, 2016-04-18:
@svdgraaf the code at github is my most up to date code... I just didn't upload the example code yet, but I've posted it above. it seems that the memory issue was fixed.

odelot, 2016-04-18:
@svdgraaf the code at github is my most up to date code... I just didn't upload the example code yet, but I've posted it above. it seems that the memory issue was fixed.

aaroneiche, 2016-08-19:
I found the sample code in the [above comment](https://github.com/odelot/aws-mqtt-websockets/issues/2#issuecomment-208978640) to be tremendously helpful. The Sample I'd been working off previously was failing frequently on connection. This sample worked great :) I am connected and successfully receiving messages.

aaroneiche, 2016-08-19:
It seems I spoke too soon - I'm still running into issues getting the websocket connection to work:
```

connected
11904 - conn: 1 - (26504)
please start sntp first !
[AWSc] Disconnected!
please start sntp first !
[AWSc] Disconnected!
please start sntp first !
pm open,type:2 0
[AWSc] Disconnected!
please start sntp first !
[AWSc] Disconnected!
please start sntp first !
error connection to the websocket server
19491 - conn: 2 - (11360)
[AWSc] Disconnected!
please start sntp first !
[AWSc] Disconnected!
please start sntp first !
[AWSc] Disconnected!
please start sntp first !
[AWSc] Disconnected!
please start sntp first !
[AWSc] Disconnected!
please start sntp first !
error connection to the websocket server
26987 - conn: 3 - (11360)
[AWSc] Disconnected!
please start sntp first !
[AWSc] Disconnected!
please start sntp first !
[AWSc] Disconnected!
please start sntp first !
[AWSc] Disconnected!
please start sntp first !
[AWSc] Disconnected!
please start sntp first !
error connection to the websocket server
34480 - conn: 4 - (11360)
```
It just keeps on rolling like that. Any suggestions on what to do to make the websocket connect more reliable? I got it working once, but now I have no idea what make that work.

jatindera, 2017-02-10:
Hi,
Till yesterday, everything was working fine for me. I connected my ESP8266 development board to AWS IoT. Today morning, I connected it again without making any change and I started getting following error:

please start sntp first !
error connection to the websocket server

Appreciate your help on this.

sh4nnongoh, 2017-02-16:
Regarding the SNTP issue:

Add the following line to the Setup section,

**configTime(3 * 3600, 0, "pool.ntp.org", "time.nist.gov");**