odelot/aws-mqtt-websockets #5: freezes esp8266 after ~40 reconnection

odelot, 2016-03-29:
probably a memory leak.

I hope to be in my library. The other option is inside the connection/desconnection procedure from paho mqtt and arduino websockets (out of my control)

odelot, 2016-03-30:
fixed memory leaks ;-) it seems that it was all in my library \o/

8h reconnecting without freeze (~301 times and counting)

odelot, 2016-03-30:
freezes after 413 reconnection.... :-(

I will look for more memory leaks in my code. any help is appreciated

Panic C:\Users\fabio-note\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\libraries\ESP8266WiFi\src\WiFiClientSecure.cpp:495 ax_port_malloc

ctx: cont 
sp: 3fff17d0 end: 3fff1c20 offset: 01b0

odelot, 2016-04-01:
It seems that my code does not have any more memory leaks

After some digging, it seems that the problem is with esp8266 arduino framework, but i am not that sure.

I've posted to the community to see if it is a known behavior http://www.esp8266.com/viewtopic.php?f=32&t=9300

odelot, 2016-04-28:
see the README.md to fix memory leak on Arduino Core for ESP8266

closed 2016-04-28 by odelot

pfabri, 2016-12-10:
Hi Odelot,
Which README.md shall we see? Could you please include a link? 
Many thanks!

odelot, 2016-12-10:
@pfabri With Arduino/ESP sdk version 2.3.0 you don't need to fix this memory leak anymore