odelot/aws-mqtt-websockets #6: not able to connect to AWS IOT

mrb0y, 2016-04-28:
Hi! Awesome job on getting the ESP8266 talking with AWS IOT. I have set up a "thing" on my AWS IOT account. I used the credentials created for a single user with AWS IOT privileges, verified the end point path, region and the like via the details output from the AWS IOS console and populate the AWS IOT config variables. I'm using a  NodeMCU 0.9(ESP-12 Module), 80MHz, 115200, 4M (3M SPIFFS) for the build option in the IDE. I'm able to get an internet connection via Wifi router but it fails when attempting to connect to AWS. Should I have some other configuration completed in the AWS IOT console? I basically have a single user with iot credentials and then used that to create a thing. Thanks for any help.

Serial console out put...
connected
error connection to the websocket server
error connection to the websocket server
 
Serial output from the more verbose memleak debug version...
error connection to the websocket server
38923 - conn: 5 - (27216)
error connection to the websocket server
46364 - conn: 6 - (27216)
error connection to the websocket server
53837 - conn: 7 - (27216)
error connection to the websocket server
61294 - conn: 8 - (27216)

daniele-salvagni, 2016-04-28:
Could you try with this example instead? https://github.com/odelot/aws-mqtt-websockets/issues/2#issuecomment-208978640

If I recall correctly I also had some problems with the current example after the most recent commits, let me know if it does solve the issue, in that case the new example should be committed to the repository.

odelot, 2016-04-28:
To help yo understand what is happening, you can turn debug messages on.

in the example file, uncomment this line:
`//Serial.setDebugOutput(1);`

in the` AWSWebSocketClient.h` file, uncomment this line:
`//#define DEBUG_WEBSOCKET_MQTT(...) os_printf( __VA_ARGS__ )`

in the `WebSockets.h` file, from `arduinoWebSockets` library, uncomment this line:
`//#define DEBUG_WEBSOCKETS(...) os_printf( __VA_ARGS__ )`

Btw, i will commit the new example file. I just was waiting some feedback about the changes. ;-)

mrb0y, 2016-04-28:
Thanks for the information. I will run the new example today. Have two NodeMCU to test on. I don't have the AWS MQTT configured properly. Can you advise on the AWS Configuration for MQTT? I watched your blog video and I have not configured for an MQTT client on AWS. :-B

mrb0y, 2016-04-29:
It's still not working. I suspect esp-aws-sdk library dependencies. I had a build that was compiling but not by using your add these libraries and AWS credentials. When I do that I have dependencies failures in the compile. The version I can get to compile will not connect too web socket.

daniele-salvagni, 2016-04-29:
The `aws-sdk-arduino` repository from @awslabs has errors and will not compile without fixing them, which one are you using?

You should set your endpoint as follows:

    char aws_endpoint[] = "A1OKD21AI4PT54.iot.us-west-2.amazonaws.com";

mrb0y, 2016-04-30:
I was using the daniele-salvagni / aws-sdk-esp8266 library to compile. I'm hoping if I can go back to using that and use the correct end point it will work. What errors need to be corrected in the @awslabs library in order to compile? Thanks again for information and timely replies. :)

shreya95, 2016-05-02:
Hi! Even I have been getting the same error and I'm using a NodeMCU ESP8266 (Amica). Initially there were some errors while compiling using the aws-sdk-arduino library from @awslabs but I fixed it. Also, I tried using a different wifi connection and it printed "please start sntp first" onto the serial monitor before "error connection to the websocket server".

odelot, 2016-05-02:
Hi, I am a bit busy but I will soon reconfigure all my environment and re-test everything.

Initially I was using aws-sdk-arduino from svdgraaf https://github.com/svdgraaf/aws-sdk-arduino then svdgraaf made some changes in the code to do not need his version. That's why I migrated to the awslabs version (I didnt know that there was compilation errors in awslabs version. svdgraaf version compiled without errors, try to use his fork).

daniele-salvagni, 2016-05-03:
I just started a new environment from scratch (I'm using PlatformIO) with up to date code from every dependency and the ESP8266 Arduino Core 2.2.0. I'm using the SDK from this fork https://github.com/daniele-salvagni/aws-sdk-esp8266 (but also the one from @svdgraaf should work correctly). The example from this repository is working without issues, here is the last part of the output:

```
connected
30744 - conn: 1 - (38312)
please start sntp first !
websocket layer connected
￼
MQTT connecting
MQTT connected
MQTT subscribed
pm open,type:2 0
```

I didn't test the latest commits from the Arduino Core though, could it be an introduced bug?

----------

@odelot: Yes, aws-sdk-arduino from @awslabs looks to be an abandoned project and the code does not even compile because of some trivial errors and a screwed up merge, that's the reason I started this fork https://github.com/daniele-salvagni/aws-sdk-esp8266 which has now been detached into a standalone repository (Pull Requests would never be accepted and the intentions of the original developers aren't clear).

Speaking of the SDK, the current code of `aws-mqtt-websockets` is currently using  `#include "AWSClient2.h"`. At the current state `AWSClient2.h` was only created as a failed attempt to reduce the size of the code because it didn't fit the Spark Core for a few bytes (failed because as I'm aware of it wasn't enough). You should consider using `AWSClient.h` instead and use constants like `HEADER_COUNT` instead of the `HEADER_COUNT2` counterpart which are the same. 

The only difference between `AWSClient.h` and `AWSClient2.h` is that the first has a few more lines of code for CURL support. I plan to remove `AWSClient2.h` from my fork and remove CURL support from `AWSClient.h`, I didn't make this change yet because I was aware of some people using my fork as a dependency for `aws-mqtt-websockets` and I didn't want to break the compatibility. I will also drop the support for the other boards and focus only the ESP8266.

mrb0y, 2016-05-06:
I started using Platform.io my self and will try to reproduce daniele's results.

mrb0y, 2016-05-12:
I deleted all libraries and reinstalled arduino 1.6.5r. I then used only the libraries referenced in this thread for the compile. I am now getting...

error connection to the websocket server
1682390 - conn: 202 - (12824)
please start sntp first !

yachnyymaxim, 2016-05-15:
Hi all.
I see the same issue on my side - I can't connect to the AWS IoT server even using https://github.com/daniele-salvagni/aws-sdk-esp8266 SDK.

I've turned on all debugging output that I could - you can find it in the attachement: [putty.txt](https://github.com/odelot/aws-mqtt-websockets/files/264994/putty.txt)

AntorFr, 2016-05-21:
Hi all,

sames issue :  

120257 - conn: 13 - (22552)
please start sntp first !
please start sntp first !
please start sntp first !
please start sntp first !
please start sntp first !
please start sntp first !
please start sntp first !
please start sntp first !
please start sntp first !
please start sntp first !
error connection to the websocket server

boraozgen, 2016-06-16:
https://github.com/odelot/aws-mqtt-websockets/issues/9

2dom, 2016-06-22:
Same problem here ... the odd thing is that it did connect at some point (same output as  daniele-salvagni from above) but now I only get the following output. It does not even send a header ... 

7341 - conn: 1 - (33328)
Connecting to server ....
[WS-Client] connect wss...
[WS-Client] connection to A1V2EBE1FOIBLQ.iot.us-east-1.amazonaws.com:443 Faild
[WS-Client] client disconnected.
[WS-Client] connect wss...
[WS-Client] connection to A1V2EBE1FOIBLQ.iot.us-east-1.amazonaws.com:443 Faild
[WS-Client] client disconnected.
[WS-Client] connect wss...
[WS-Client] connection to A1V2EBE1FOIBLQ.iot.us-east-1.amazonaws.com:443 Faild
...

svdgraaf, 2016-06-22:
Interestingly, I still have a build combination locally that works, still at the same versions that were mentioned before (havent updated since)

Regards,
Sander

boraozgen, 2016-06-22:
@2dom check thread #9 
Latest commit of arduinoWebSockets library does not work.

2dom, 2016-06-23:
yes ... I tried all available versions of the arduinoWebSockets library ... 

I started up the same code from yeserday and it works! So either AWS has some mechanism to limit the IoT connection count or this is a time-of-the-day kind of bug,

Successful connect debug output for completeness:

connected with DWR-921-80E2, channel 11
dhcp client start...
..ip:192.168.0.73,mask:255.255.255.0,gw:192.168.0.1
.WiFi connected
IP address: 
192.168.0.73
Internet connection ok
ping 5, timeout 0, total payload 160 bytes, 4938 ms
6733 - conn: 1 - (33352)
Connecting to server ....
[WS-Client] connect wss...
please start sntp first !
[WS-Client] connected to A1V2ADASD.iot.us-east-1.amazonaws.com:443.
[WS-Client][sendHeader] sending header...
[WS-Client][sendHeader] sending header... Done (37845us).
[WS-Client][handleHeader] RX: HTTP/1.1 101 Switching Protocols
[WS-Client][handleHeader] RX: content-length: 0
[WS-Client][handleHeader] RX: upgrade: websocket
[WS-Client][handleHeader] RX: connection: Upgrade
[WS-Client][handleHeader] RX: sec-websocket-accept: 1MNPy0iiqkMctb42/i62FvdNDGE=
[WS-Client][handleHeader] RX: sec-websocket-protocol: mqtt
[WS-Client][handleHeader] Header read fin.
[WS-Client][handleHeader] Client settings:
[WS-Client][handleHeader]  - cURL: /mqtt?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=ASJDVSDASDASVW6WCQ%2F20160623%2Fus-east-1%2Fiotdevicegateway%2Faws4_request&X-Amz-Date=20160623T054301Z&X-Amz-Expires=86400&X-Amz-SignedHeaders=host&X-Amz-Signature=38f68adASDASDASDASDa50c12b919AASDASDASASDdasdsad3a84fc976a
[WS-Client][handleHeader]  - cKey: 56W3DonacXmrfhuZoKoQcA==
[WS-Client][handleHeader] Server header:
[WS-Client][handleHeader]  - cCode: 101
[WS-Client][handleHeader]  - cIsUpgrade: 1
[WS-Client][handleHeader]  - cIsWebsocket: 1
[WS-Client][handleHeader]  - cAccept: 1MNPy0iiqkMctb42/i62FvdNDGE=
[WS-Client][handleHeader]  - cProtocol: mqtt
[WS-Client][handleHeader]  - cExtensions: 
[WS-Client][handleHeader]  - cVersion: 0
[WS-Client][handleHeader] Websocket connection init done.
[WS][0][headerDone] Header Handling Done (0us).
websocket layer connected
MQTT connecting
[WS][0][sendFrame] ------- send massage frame -------
[WS][0][sendFrame] fin: 1 opCode: 2 mask: 1 length: 38 heapm open,type:2 0
derToPayload: 0
[WS][0][sendFrame] pack to one TCP package...
[WS][0][sendFrame] sending Frame Done (51011us).
[WS][0][handleWebsocketWaitFor] size: 2 cWsRXsize: 0
[readCb] n: 2 t: 10909
[WS][0][handleWebsocketWaitFor][readCb] size: 2 ok: 1
[WS][0][handleWebsocket] ------- read massage frame -------
[WS][0][handleWebsocket] fin: 1 rsv1: 0 rsv2: 0 rsv3 0  opCode: 2
[WS][0][handleWebsocket] mask: 0 payloadLen: 4
[readCb] n: 4 t: 11092
MQTT connected
[WS][0][sendFrame] ------- send massage frame -------
[WS][0][sendFrame] fin: 1 opCode: 2 mask: 1 length: 46 headerToPayload: 0
[WS][0][sendFrame] pack to one TCP package...
[WS][0][sendFrame] sending Frame Done (62264us).
[WS][0][handleWebsocketWaitFor] size: 2 cWsRXsize: 0
[readCb] n: 2 t: 11619
[WS][0][handleWebsocketWaitFor][readCb] size: 2 ok: 1
[WS][0][handleWebsocket] ------- read massage frame -------
[WS][0][handleWebsocket] fin: 1 rsv1: 0 rsv2: 0 rsv3 0  opCode: 2
[WS][0][handleWebsocket] mask: 0 payloadLen: 5
[readCb] n: 5 t: 11801
MQTT subscribed
[WS][0][sendFrame] ------- send massage frame -------
[WS][0][sendFrame] fin: 1 opCode: 2 mask: 1 length: 105 headerToPayload: 0
[WS][0][sendFrame] pack to one TCP package...
[WS][0][sendFrame] sending Frame Done (49434us).
[WS][0][handleWebsocketWaitFor] size: 2 cWsRXsize: 0
[readCb] n: 2 t: 12208
[WS][0][handleWebsocketWaitFor][readCb] size: 2 ok: 1
[WS][0][handleWebsocket] ------- read massage frame -------
[WS][0][handleWebsocket] fin: 1 rsv1: 0 rsv2: 0 rsv3 0  opCode: 2
[WS][0][handleWebsocket] mask: 0 payloadLen: 105
[readCb] n: 105 t: 12394

rahu2581, 2016-07-04:
See https://github.com/odelot/aws-mqtt-websockets/issues/7#issuecomment-230356696

aaroneiche, 2016-08-22:
I've tried desperately to get this to work, but continue to run into the same problem as listed above, with AWS returning a 403.
At this point, I don't really know how to debug. From what I can tell, the issue with the port in the header has been fixed.

I had one successful connection but haven't been able to replicate it, nor determine why. Is there a good next-step if I'm running into the 403 error? I'm confident that my endpoint, key, secret, and region are correct.

My configuration:
```
//AWS IOT config, change these:
char wifi_ssid[]       = "TARDIS";
char wifi_password[]   = "*********";
char aws_endpoint[]    = "A1IB15BSHGYGBU.iot.us-east-1.amazonaws.com";
char aws_key[]         = "AAAAAAAAAAAAAAAAA";
char aws_secret[]      = "*****************";
char aws_region[]      = "us-east-1";
const char* aws_topic  = "$aws/things/Smartish_Device_1/shadow/update";
int port = 443;

```

The Serial Output
```
connected with TARDIS, channel 1
dhcp client start...
ip:192.168.1.114,mask:255.255.255.0,gw:192.168.1.1

connected
11861 - conn: 1 - (30288)
[WS-Client] connect wss...
please start sntp first !
[WS-Client] connected to A1IB15BSHGYGBU.iot.us-east-1.amazonaws.com:443.
[WS-Client][sendHeader] sending header...
[WS-Client][sendHeader] sending header... Done (105636us).
[WS-Client][handleHeader] RX: HTTP/1.1 403 Forbidden
[WS-Client][handleHeader] RX: content-type: application/json
[WS-Client][handleHeader] RX: content-length: 241
[WS-Client][handleHeader] RX: date: Mon, 22 Aug 2016 05:18:35 GMT
[WS-Client][handleHeader] RX: x-amzn-RequestId: 1d31bcab-12b9-73e4-8d73-8a8bdd0500bc
[WS-Client][handleHeader] RX: connection: Keep-Alive
[WS-Client][handleHeader] RX: x-amzn-ErrorType: ForbiddenException:
[WS-Client][handleHeader] Header read fin.
[WS-Client][handleHeader] Client settings:
[WS-Client][handleHeader]  - cURL: /mqtt?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAJTZXUCKBZGHC7VSA%2F20160822%2Fus-east-1%2Fiotdevicegateway%2Faws4_request&X-Amz-Date=20160822T051833Z&X-Amz-Expires=86400&X-Amz-SignedHeaders=host&X-Amz-Signature=ccb606feff3b688b97450ef2c11182a16e8288a172cae5543c7f586e4bd068ac
[WS-Client][handleHeader]  - cKey: hz3kkN4UtdzG2prSUGTPZg==
[WS-Client][handleHeader] Server header:
[WS-Client][handleHeader]  - cCode: 403
[WS-Client][handleHeader]  - cIsUpgrade: 0
[WS-Client][handleHeader]  - cIsWebsocket: 1
[WS-Client][handleHeader]  - cAccept: 
[WS-Client][handleHeader]  - cProtocol: mqtt
[WS-Client][handleHeader]  - cExtensions: 
[WS-Client][handleHeader]  - cVersion: 0
[WS-Client][handleHeader] no Websocket connection close.
[WS-Client] client disconnected.
[AWSc] Disconnected!
[WS-Client] connect wss...
please start sntp first !
```

rahu2581, 2016-08-22:
Two things that might help:
`please start sntp first !`
Set your time first
`char aws_endpoint[]    = "A1IB15BSHGYGBU.iot.us-east-1.amazonaws.com";`
Try the host name in lower-case

aaroneiche, 2016-08-22:
> Set your time first
How do I do this?

As for the endpoint case name, I was trying upper-case as @2dom had success with upper. I've switched it back since. One thing that I did note was that the version of ArduinoWebSockets that I got from the Arduino Library manager was older, and did not have the port appended to the host in the handshake. I've pulled down a more recent copy and am trying to compile it (it looks like there's a signature mismatch on one of the methods)

vitoralvimb, 2016-10-25:
Hi folks!
I have the same problem that aaroneiche.

Anyone can help me with this problem? All keys are right and I couldn´t connect.

9794 - conn: 1 - (31184)
[WS-Client] connect wss...
[WS-Client] connected to ayp36v055qz3m.iot.us-west-2.amazonaws.com:443.
[WS-Client][sendHeader] sending header...
[WS-Client][sendHeader] sending header... Done (261793us).
[WS-Client][handleHeader] RX: HTTP/1.1 403 Forbidden
[WS-Client][handleHeader] RX: content-type: application/json
[WS-Client][handleHeader] RX: content-length: 241
[WS-Client][handleHeader] RX: date: Tue, 25 Oct 2016 00:13:11 GMT
[WS-Client][handleHeader] RX: x-amzn-RequestId: f94c7bcb-f028-4c13-77f0-afccb8a01947
[WS-Client][handleHeader] RX: connection: Keep-Alive
[WS-Client][handleHeader] RX: x-amzn-ErrorType: ForbiddenException:
[WS-Client][handleHeader] RX: access-control-allow-origin: *
[WS-Client][handleHeader] RX: access-control-allow-headers: Authorization
[WS-Client][handleHeader] RX: access-control-allow-headers: X-amz-security-token
[WS-Client][handleHeader] RX: access-control-allow-headers: Accept
[WS-Client][handleHeader] RX: access-control-allow-headers: X-amz-date
[WS-Client][handleHeader] RX: access-control-allow-headers: X-amz-user-agent
[WS-Client][handleHeader] RX: access-control-allow-headers: X-amz-content-sha256
[WS-Client][handleHeader] RX: access-control-allow-headers: Accept-Language
[WS-Client][handleHeader] RX: access-control-allow-headers: X-requested-with
[WS-Client][handleHeader] RX: access-control-allow-headers: Content-Language
[WS-Client][handleHeader] RX: access-control-allow-headers: Content-Type
[WS-Client][handleHeader] RX: access-control-expose-headers: x-amzn-ErrorMessage
[WS-Client][handleHeader] RX: access-control-expose-headers: x-amzn-RequestId
[WS-Client][handleHeader] RX: access-control-expose-headers: x-amzn-ErrorType
[WS-Client][handleHeader] RX: access-control-expose-headers: Date
[WS-Client][handleHeader] Header read fin.
[WS-Client][handleHeader] Client settings:
[WS-Client][handleHeader]  - cURL: /mqtt?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=<my credential goes here OK>%2F20161025%2Fus-west-2%2Fiotdevicegateway%2Faws4_request&X-Amz-Date=20161025T001307Z&X-Amz-Expires=86400&X-Amz-SignedHeaders=host&X-Amz-Signature=8445837a9a9b5b787279e1842ddc46dfc0cc43669b0b515fc84c6268ed69ace8
[WS-Client][handleHeader]  - cKey: pCd37rrxnpuKKwI3aOlJSQ==
[WS-Client][handleHeader] Server header:
[WS-Client][handleHeader]  - cCode: 403
[WS-Client][handleHeader]  - cIsUpgrade: 0
[WS-Client][handleHeader]  - cIsWebsocket: 1
[WS-Client][handleHeader]  - cAccept: 
[WS-Client][handleHeader]  - cProtocol: mqtt
[WS-Client][handleHeader]  - cExtensions: 
[WS-Client][handleHeader]  - cVersion: 0
[WS-Client][handleHeader]  - cSessionId: 
[WS-Client][handleHeader] no Websocket connection close.
[WS-Client] client disconnected.

odelot, 2016-11-24:
I've fixed the library to work with the latest version of websockets library (2.0.5) and arduino/esp sdk (2.3.0)

use the endpoint in lowercase 

hope it works for you now ;-)

theTechGrandma, 2016-12-27:
Hey there! I'm was working with this code and had the same issues. I'm off this week and opened it up to tackle it. Glad to see the update. I'll give it a shot and let you know!

theTechGrandma, 2016-12-27:
Super happy! It works!! I removed all the previous libraries. Reset up the new libraries and TA-DA! Connected the first time.

ashwnaipal, 2017-02-08:
please help me on connecting with aws iot services with esp8266 12e. i am using example aws-mqtt-websocket example not able to connect with dashboard of aws iot the code is as follow and the error i get is (please start sntp first ! error connection to the websocket server)

`#include <Arduino.h>
  #include <Stream.h>

  #include <ESP8266WiFi.h>

  //AWS
  #include "sha256.h"
  #include "Utils.h"
  #include "AWSClient2.h"

  //WEBSockets
  #include <Hash.h>
  #include <WebSocketsClient.h>

  //MQTT PAHO
  #include <SPI.h>
  #include <IPStack.h>
  #include <Countdown.h>
  #include <MQTTClient.h>

  //AWS MQTT Websocket
  #include "Client.h"
  #include "AWSWebSocketClient.h"
  #include "CircularByteBuffer.h"

  //AWS IOT config, change these:

  const char* ssid = "NPSE";
  const char* password = "P/W070415";
  char aws_endpoint[]    = "aph0r6gk1etz5.iot.us-east-1.amazonaws.com";
  char aws_key[]         = "AKIAJDJA6HPCGZXG7N6Q";
  char aws_secret[]      = "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";
  char aws_region[]      = "us-east-1";
  const char* aws_topic  = "$aws/things/tutorial/shadow/update";
  int port = 443;

  //MQTT config
  const int maxMQTTpackageSize = 512;
  const int maxMQTTMessageHandlers = 1;

  WiFiClientSecure espclient;

  AWSWebSocketClient awsWSclient(1000);

  IPStack ipstack(awsWSclient);

  MQTT::Client<IPStack, Countdown, maxMQTTpackageSize, maxMQTTMessageHandlers> *client = NULL;

  //# of connections
  long connection = 0;

  //generate random mqtt clientID
  char* generateClientID () {
    char* cID = new char[23]();
    for (int i=0; i<22; i+=1)
      cID[i]=(char)random(1, 256);
    return cID;
  }

  //count messages arrived
  int arrivedcount = 0;

  //callback to handle mqtt messages
  void messageArrived(MQTT::MessageData& md)
  {
    MQTT::Message &message = md.message;

    Serial.print("Message ");
    Serial.print(++arrivedcount);
    Serial.print(" arrived: qos ");
    Serial.print(message.qos);
    Serial.print(", retained ");
    Serial.print(message.retained);
    Serial.print(", dup ");
    Serial.print(message.dup);
    Serial.print(", packetid ");
    Serial.println(message.id);
    Serial.print("Payload ");
    char* msg = new char[message.payloadlen+1]();
    memcpy (msg,message.payload,message.payloadlen);
    Serial.println(msg);
    delete msg;
  }

  //connects to websocket layer and mqtt layer
  bool connect () {

      if (client == NULL) {
        client = new MQTT::Client<IPStack, Countdown, maxMQTTpackageSize, maxMQTTMessageHandlers>(ipstack);
      } else {

        if (client->isConnected ()) {    
          client->disconnect ();
        }  
        delete client;
        client = new MQTT::Client<IPStack, Countdown, maxMQTTpackageSize, maxMQTTMessageHandlers>(ipstack);
      }

      //delay is not necessary... it just help us to get a "trustful" heap space value
      delay (1000);
      Serial.print (millis ());
      Serial.print (" - conn: ");
      Serial.print (++connection);
      Serial.print (" - (");
      Serial.print (ESP.getFreeHeap ());
      Serial.println (")");

     int rc = ipstack.connect(aws_endpoint, port);
     Serial.print(rc);

      if (rc != 1)
      {
        Serial.println("error connection to the websocket server");
        return false;
      } else {
        Serial.println("websocket layer connected");
      }

      Serial.println("MQTT connecting");
      MQTTPacket_connectData data = MQTTPacket_connectData_initializer;
      data.MQTTVersion = 3;
      char* clientID = generateClientID ();
      data.clientID.cstring = clientID;
      rc = client->connect(data);
      delete[] clientID;
      if (rc != 0)
      {
        Serial.print("error connection to MQTT server");
        Serial.println(rc);
        return false;
      }
      Serial.println("MQTT connected");
      return true;
  }

  //subscribe to a mqtt topic
  void subscribe () {
     //subscript to a topic
      int rc = client->subscribe(aws_topic, MQTT::QOS0, messageArrived);
      Serial.println(rc);
      if (rc != 0) {
        Serial.print("rc from MQTT subscribe is ");
        Serial.println(rc);
        return;
      }
      Serial.println("MQTT subscribed");
  }

  //send a message to a mqtt topic
  void sendmessage () {
      //send a message
      MQTT::Message message;
      char buf[100];
      strcpy(buf, "{\"state\":{\"reported\":{\"on\": false}, \"desired\":{\"on\": false}}}");
      message.qos = MQTT::QOS0;
      message.retained = false;
      message.dup = false;
      message.payload = (void*)buf;
      message.payloadlen = strlen(buf)+1;
      int rc = client->publish(aws_topic, message); 
  }

  void setup_wifi() {

    delay(10);
    // We start by connecting to a WiFi network
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(ssid);

    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
  }

  void setup() {
      Serial.begin (115200);
      setup_wifi();

      Serial.setDebugOutput(1); 
     //fill AWS parameters    
      awsWSclient.setAWSRegion(aws_region);
      awsWSclient.setAWSDomain(aws_endpoint);
      awsWSclient.setAWSKeyID(aws_key);
      awsWSclient.setAWSSecretKey(aws_secret);
      awsWSclient.setUseSSL(true);

      if (connect ()){
        subscribe ();
        sendmessage ();
      }

  }

  void loop() {
    //keep the mqtt up and running
    if (awsWSclient.connected ()) {    
        client->yield();
    } else {
      //handle reconnection
      if (connect ()){
        subscribe ();      
      }
    }

  }

[image]

joeinaustin, 2017-02-09:
After following multiple references to this example, I was able to get the basics to work.  However, I am also seeing the "please start sntp first !" message on connection.  Someone above suggested to "set the time first" but gave no details.  The ESP8266 has no rtc, so perhaps this message should simply be ignored????  Nor do not want to load up the 8266 with RTC code.  It's necessary for my app.

Has anyone worked out the source of this message?

odelot, 2017-02-09:
@joeinaustin I think the "please start sntp first" is a log from the ESP8266 arduino SDK. You can turn the debug off changing this line `Serial.setDebugOutput(1);`  to `Serial.setDebugOutput(0);`

I think ESP has some kind of RTC emulation, someone can correct me if I am wrong. There are also some functions to use the RTC internal memory to store data between deep sleep restarts.  See `ESP.rtcUserMemoryWrite` and `ESP.rtcUserMemoryRead` here: http://esp8266.github.io/Arduino/versions/2.3.0/doc/libraries.html#esp-specific-apis

sh4nnongoh, 2017-02-16:
Add this line in setup to solve the SNTP issue:

 **configTime(3 * 3600, 0, "pool.ntp.org", "time.nist.gov");**

sh4nnongoh, 2017-02-16:
I am still having the Http 403 Forbidden issue during the initial connection :/

I am using the appropriate versions of the various libraries.

ls6777, 2017-03-07:
I'm also getting the 403 Forbidden issue during connection to the websocket.  Below is my debug output.  I verified my keys are right, and I can connect to my "thing" using a AWS generated Python script that's doing MQTT over websockets.  I've been trying to get this thing working for 2 weeks with no success and I've combed through all of these comments/forums/suggestions.  Any help would be much appreciated!!

Here is my configuration:
```
char wifi_ssid[]       = "8A1256";
char wifi_password[]   = "******************";
char aws_endpoint[]    = "a2s21egpq7sdet.iot.us-west-2.amazonaws.com";
char aws_key[]         = "***************************";
char aws_secret[]      = "*******************************************";
char aws_region[]      = "eu-west-2";
const char* aws_topic  = "$aws/things/TestDevice/shadow/update";
int port = 443;

//MQTT config
const int maxMQTTpackageSize = 1024;
const int maxMQTTMessageHandlers = 1;
```

Here is the output:
```
33625 - conn: 1 - (31648)<\r><\n>
[WS-Client] connect wss...<\n>
please start sntp first !<\n>
[WS-Client] connected to a2s21egpq7sdet.iot.us-west-2.amazonaws.com:443.<\n>
[WS-Client][sendHeader] sending header...<\n>
[WS-Client][sendHeader] sending header... Done (64638us).<\n>
[WS-Client][handleHeader] RX: HTTP/1.1 403 Forbidden<\n>
[WS-Client][handleHeader] RX: content-type: application/json<\n>
[WS-Client][handleHeader] RX: content-length: 127<\n>
[WS-Client][handleHeader] RX: date: Tue, 07 Mar 2017 03:07:35 GMT<\n>
[WS-Client][handleHeader] RX: x-amzn-RequestId: 5e6ec588-9316-3c06-2899-6c080285e2a9<\n>
[WS-Client][handleHeader] RX: x-amzn-ErrorType: ForbiddenException:<\n>
[WS-Client][handleHeader] RX: access-control-allow-origin: *<\n>
[WS-Client][handleHeader] RX: access-control-allow-headers: Authorization<\n>
[WS-Client][handleHeader] RX: access-control-allow-headers: X-amz-security-token<\n>
[WS-Client][handleHeader] RX: access-control-allow-headers: Accept<\n>
[WS-Client][handleHeader] RX: access-control-allow-headers: X-amz-date<\n>
[WS-Client][handleHeader] RX: access-control-allow-headers: X-amz-user-agent<\n>
[WS-Client][handleHeader] RX: access-control-allow-headers: X-amz-content-sha256<\n>
[WS-Client][handleHeader] RX: access-control-allow-headers: Accept-Language<\n>
[WS-Client][handleHeader] RX: access-control-allow-headers: X-requested-with<\n>
[WS-Client][handleHeader] RX: access-control-allow-headers: Content-Language<\n>
[WS-Client][handleHeader] RX: access-control-allow-headers: Content-Type<\n>
[WS-Client][handleHeader] RX: access-control-expose-headers: x-amzn-ErrorMessage<\n>
[WS-Client][handleHeader] RX: access-control-expose-headers: x-amzn-RequestId<\n>
[WS-Client][handleHeader] RX: access-control-expose-headers: x-amzn-ErrorType<\n>
[WS-Client][handleHeader] RX: access-control-expose-headers: Date<\n>
[WS-Client][handleHeader] Header read fin.<\n>
[WS-Client][handleHeader] Client settings:<\n>
[WS-Client][handleHeader]  - cURL: /mqtt?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAJDHNR5HNTY5A2QOQ%2F20170307%2Feu-west-2%2Fiotdevicegateway%2Faws4_request&X-Amz-Date=20170307T030733Z&X-Amz-Expires=86400&X-Amz-SignedHeaders=host&X-Amz-Signature=56b388d5c633ff6d85361b6d80a6d80235641e4a815545245c4691f229538c81<\n>
[WS-Client][handleHeader]  - cKey: 7r5tvbpDxLqRemtfqdoVmw==<\n>
[WS-Client][handleHeader] Server header:<\n>
[WS-Client][handleHeader]  - cCode: 403<\n>
[WS-Client][handleHeader]  - cIsUpgrade: 0<\n>
[WS-Client][handleHeader]  - cIsWebsocket: 1<\n>
[WS-Client][handleHeader]  - cAccept: <\n>
[WS-Client][handleHeader]  - cProtocol: mqtt<\n>
[WS-Client][handleHeader]  - cExtensions: <\n>
[WS-Client][handleHeader]  - cVersion: 0<\n>
[WS-Client][handleHeader]  - cSessionId: <\n>
[WS-Client][handleHeader] no Websocket connection close.<\n>
[WS-Client] client disconnected.<\n>
[AWSc] Disconnected!<\n>
[WS-Client] connect wss...<\n>
please start sntp first !<\n>
[WS-Client] connected to a2s21egpq7sdet.iot.us-west-2.amazonaws.com:443.<\n>
[WS-Client][sendHeader] sending header...<\n>
[WS-Client][sendHeader] sending header... Done (64762us).<\n>
[WS-Client][handleHeader] RX: HTTP/1.1 403 Forbidden<\n>
[WS-Client][handleHeader] RX: content-type: application/json<\n>
[... 11 lines cut ...]
[WS-Client][handleHeader] RX: access-control-allow-headers: Accept-Language<\n>
[WS-Client][handleHeader] RX: access-control-allow-headers: X-requested-with<\n>
[WS-Client][handleHeader] RX: access-control-allow-headers: Content-Language<\n>
[WS-Client][handleHeader] RX: access-control-allow-headers: Content-Type<\n>
[WS-Client][handleHeader] RX: access-control-expose-headers: x-amzn-ErrorMessage<\n>
[WS-Client][handleHeader] RX: access-control-expose-headers: x-amzn-RequestId<\n>
[WS-Client][handleHeader] RX: access-control-expose-headers: x-amzn-ErrorType<\n>
[WS-Client][handleHeader] RX: access-control-expose-headers: Date<\n>
[WS-Client][handleHeader] Header read fin.<\n>
[WS-Client][handleHeader] Client settings:<\n>
[WS-Client][handleHeader]  - cURL: /mqtt?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAJDHNR5HNTY5A2QOQ%2F20170307%2Feu-west-2%2Fiotdevicegateway%2Faws4_request&X-Amz-Date=20170307T030733Z&X-Amz-Expires=86400&X-Amz-SignedHeaders=host&X-Amz-Signature=56b388d5c633ff6d85361b6d80a6d80235641e4a815545245c4691f229538c81<\n>
[WS-Client][handleHeader]  - cKey: AvsH72VHTLutsKzBrjSQAQ==<\n>
[WS-Client][handleHeader] Server header:<\n>
[WS-Client][handleHeader]  - cCode: 403<\n>
[WS-Client][handleHeader]  - cIsUpgrade: 0<\n>
[WS-Client][handleHeader]  - cIsWebsocket: 0<\n>
[WS-Client][handleHeader]  - cAccept: <\n>
[WS-Client][handleHeader]  - cProtocol: <\n>
[WS-Client][handleHeader]  - cExtensions: <\n>
[WS-Client][handleHeader]  - cVersion: 0<\n>
[WS-Client][handleHeader]  - cSessionId: <\n>
[WS-Client][handleHeader] no Websocket connection close.<\n>
[WS-Client] client disconnected.<\n>
[AWSc] Disconnected!<\n>
```

jrkosinski, 2017-03-28:
I am having the same exact issue as ls6777. Any luck?

teejK, 2017-07-30:
@jrkosinski @ls6777 @sh4nnongoh 

spent all day pulling my hair out on this.. should have tried the fix in a previously closed issue

https://github.com/odelot/aws-mqtt-websockets/issues/7#issuecomment-259487530

IT WORKS!!!!

[image]

using arduinoWebSockets v.2.0.5 and arduino/esp sdk 2.3.0

dielume, 2017-08-24:
@teejK I followed all your previous instructions, Using:

arduino/esp sdk 2.3.0
aws-mqtt-websockets 1.0.1 
arduinoWebSockets v.2.0.5 
https://github.com/odelot/aws-mqtt-websockets/issues/7#issuecomment-259487530

Finally i can connect with AWS. However, i can't connect to MQTT this is my Serial OUTPUT

⸮connecting to wifi

connected
Thu Aug 24 21:19:54 2017

10738 - conn: 1 - (33416)
[WS-Client] connect wss...
[WS-Client] connected to a2sou3m7k9yota.iot.us-west-2.amazonaws.com:443.
[WS-Client][sendHeader] sending header...
[WS-Client][sendHeader] Header: 
GET /mqtt?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAJBI4IZB7C4U7HT7Q%2F20170824%2Fus-west-2%2Fiotdevicegateway%2Faws4_request&X-Amz-Date=20170824T181956Z&X-Amz-Expires=86400&X-Amz-SignedHeaders=host&X-Amz-Signature=5290384cc9b23da6815590aaf2a34701996c78a2704db4146e34096e79252e03 HTTP/1.1
Host: a2sou3m7k9yota.iot.us-west-2.amazonaws.com
Connection: Upgrade
Upgrade: websocket
Origin: file://
User-Agent: arduino-WebSocket-Client
Sec-WebSocket-Version: 13
Sec-WebSocket-Key: jj8lnTKhKt0T+uPE7MmgrA==
Sec-WebSocket-Protocol: mqtt

[WS-Client][sendHeader] sending header... Done (208309us).
[WS-Client][handleHeader] RX: HTTP/1.1 101 Switching Protocols
[WS-Client][handleHeader] RX: content-length: 0
[WS-Client][handleHeader] RX: upgrade: websocket
[WS-Client][handleHeader] RX: connection: upgrade
[WS-Client][handleHeader] RX: sec-websocket-accept: U45qYvVkF7n88NHFuB8ygYL3/mg=
[WS-Client][handleHeader] RX: sec-websocket-protocol: mqtt
[WS-Client][handleHeader] RX: access-control-allow-origin: *
[WS-Client][handleHeader] RX: access-control-expose-headers: x-amzn-ErrorMessage
[WS-Client][handleHeader] RX: access-control-expose-headers: x-amzn-RequestId
[WS-Client][handleHeader] RX: access-control-expose-headers: x-amzn-ErrorType
[WS-Client][handleHeader] RX: access-control-expose-headers: Date
[WS-Client][handleHeader] Header read fin.
[WS-Client][handleHeader] Client settings:
[WS-Client][handleHeader]  - cURL: /mqtt?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAJBI4IZB7C4U7HT7Q%2F20170824%2Fus-west-2%2Fiotdevicegateway%2Faws4_request&X-Amz-Date=20170824T181956Z&X-Amz-Expires=86400&X-Amz-SignedHeaders=host&X-Amz-Signature=5290384cc9b23da6815590aaf2a34701996c78a2704db4146e34096e79252e03
[WS-Client][handleHeader]  - cKey: jj8lnTKhKt0T+uPE7MmgrA==
[WS-Client][handleHeader] Server header:
[WS-Client][handleHeader]  - cCode: 101
[WS-Client][handleHeader]  - cIsUpgrade: 1
[WS-Client][handleHeader]  - cIsWebsocket: 1
[WS-Client][handleHeader]  - cAccept: U45qYvVkF7n88NHFuB8ygYL3/mg=
[WS-Client][handleHeader]  - cProtocol: mqtt
[WS-Client][handleHeader]  - cExtensions: 
[WS-Client][handleHeader]  - cVersion: 0
[WS-Client][handleHeader]  - cSessionId: 
[WS-Client][handleHeader] Websocket connection init done.
[WS][0][headerDone] Header Handling Done (0us).
[AWSc] Connected to url: /mqtt?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAJBI4IZB7C4U7HT7Q%2F20170824%2Fus-west-2%2Fiotdevicegateway%2Faws4_request&X-Amz-Date=20170824T181956Z&X-Amz-Expires=86400&X-Amz-SignedHeaders=host&X-Amz-Signature=5290384cc9b23da6815590aaf2a34701996c78a2704db4146e34096e79252e03
websocket layer connected
MQTT connecting
[WS][0][sendFrame] ------- send massage frame -------
[WS][0][sendFrame] fin: 1 opCode: 2 mask: 1 length: 29 headerToPayload: 0
[WS][0][sendFrame] pack to one TCP package...
[WS][0][sendFrame] sending Frame Done (160887us).
[WS][0][handleWebsocketWaitFor] size: 2 cWsRXsize: 0
[readCb] n: 2 t: 14673
[WS][0][handleWebsocketWaitFor][readCb] size: 2 ok: 1
[WS-Client] connection lost.
[WS-Client] client disconnected.
[AWSc] Disconnected!
[WS-Client] connect wss...
[WS-Client] connected to a2sou3m7k9yota.iot.us-west-2.amazonaws.com:443.
[WS-Client][sendHeader] sending header...
[WS-Client][sendHeader] Header: 
GET /mqtt?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAJBI4IZB7C4U7HT7Q%2F20170824%2Fus-west-2%2Fiotdevicegateway%2Faws4_request&X-Amz-Date=20170824T181956Z&X-Amz-Expires=86400&X-Amz-SignedHeaders=host&X-Amz-Signature=5290384cc9b23da6815590aaf2a34701996c78a2704db4146e34096e79252e03 HTTP/1.1
Host: a2sou3m7k9yota.iot.us-west-2.amazonaws.com
Connection: Upgrade
Upgrade: websocket
Origin: file://
User-Agent: arduino-WebSocket-Client
Sec-WebSocket-Version: 13
Sec-WebSocket-Key: /kDWA7jJ++vvlapWja8w8Q==

[WS-Client][sendHeader] sending header... Done (236767us).
error connection to MQTT server-1

My board is an Wemos D1 mini Pro. Can you help me?
Thanks

teejK, 2017-08-24:
@dielume looks like probably IAM permission issues, but you should check what cloudwatch has to say

[image]

btw if you're seting up a subscribe the docs look out of date, you should use a topic rather than topicfilter in you IAM role... or just add both.

This is my IAM for the role... but you should see if it meets your needs 

{
    "Version": "2012-10-17",
    "Statement": [
        {
            "Action": "iot:Publish",
            "Resource": [
                "[iotEndPoint]:topicfilter/$aws/things/[thingName]/shadow/*"
                "[iotEndPoint]:topic/$aws/things/[thingType]/[thingName]/shadow/*"
            ],
            "Effect": "Allow"
        },
        {
            "Action": [
                "iot:Subscribe",
                "iot:Receive"
            ],
            "Resource": [
                 "[iotEndPoint]:topicfilter/$aws/things/[thingName]/shadow/*"
                "[iotEndPoint]:topic/$aws/things/[thingName]/shadow/*"
            ],
            "Effect": "Allow"
        },
        {
            "Action": "iot:Connect",
            "Resource": "[iotEndpoint]:client/[customClientID]",
            "Effect": "Allow"
        },
        {
            "Action": "iot:DescribeEndpoint",
            "Effect": "Allow",
            "Resource": "*"
        }
    ]
}

dielume, 2017-08-28:
@teejK Thanks for your help,
My error was the policy, the configuration was bad, Now all work very well.
Too grateful, for your advices.

teejK, 2017-08-31:
It looks like the new commit (3caa6da) is working with the current version of arduinoWebSockets (1bd4638)

closed 2018-04-15 by odelot