odelot/aws-mqtt-websockets #9: Does not connect with current arduinoWebSockets version

boraozgen, 2016-06-16:
Hi,
I had the connection error issue with 403 Forbidden message. It turned out there was an error in the header string caused by the latest commits of Links2004's arduinoWebSockets library. By checking out an older version of that dependency (b3efb31) I managed to solve the problem. Maybe we can fix this somehow.
Best,

rahu2581, 2016-07-04:
The port number needs to be added before signing the header. See https://github.com/odelot/aws-mqtt-websockets/issues/7#issuecomment-230356696

tejaswigowda, 2016-07-06:
It still does not work. Rahul your fok crahes during execution (compiles fine). The strangest thing was that this worked just a couple hours ago. No sure if it was my Arduino IDE (auto-)updating WebSockets library.

musalih, 2016-07-07:
50110 - conn: 1 - (29264)

Exception (28):
epc1=0x40208f4f epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000028 depc=0x00000000

ctx: cont 
sp: 3fff1f60 end: 3fff2380 offset: 01a0

>>>stack>>>
3fff2100:  3ffef940 00000050 3fff10d0 40208f49  
3fff2110:  3fff0fd8 00000000 3ffea048 811fef36  
3fff2120:  3ffe9c8c 3ffef970 00000002 40212f68  
3fff2130:  3fff3b3c 3fff60a4 3fff10d0 402113ee  
3fff2140:  00002f80 3fff58c8 00000002 001002b7  
3fff2150:  401045ad 3ffef970 3ffef0c8 00000000  
3fff2160:  0000007f 00000030 00000008 ffffffff  
3fff2170:  40104868 00080000 7fffffff 00000080  
3fff2180:  40102a6f 00000084 3fff2cb8 00000588  
3fff2190:  00000000 3fff57e4 50600000 4000050c  
3fff21a0:  3fffc278 401028f0 3fffc200 00000022  
3fff21b0:  3fff21c0 40103abd 00040000 00000022  
3fff21c0:  40000f83 3fff6184 0000000f 00000000  
3fff21d0:  401004ba 00000004 3fff223f 3ffe83da  
3fff21e0:  000001bb 00000000 3fff0fd8 4021160b  
3fff21f0:  00000032 0000000a 3fff1294 40210d0d  
3fff2200:  3fff223b 0000000a 3fff1294 40210d0d  
3fff2210:  3fff224b 3fff22e0 3fff223a 40212231  
3fff2220:  00000006 00000194 3fff224a 40212231  
3fff2230:  00000000 00000001 3fff0fd8 40214738  
3fff2240:  3fff0000 3fff1008 3fff0fe8 40210770  
3fff2250:  3fff0fd0 3fff1294 3fff0fd8 3ffe83da  
3fff2260:  3fff0fd0 00000000 3fff0fd8 40211b3e  
3fff2270:  00000000 000001bb 3fff1294 40212120  
3fff2280:  40211bf0 3fff0fd4 3fff1294 3ffe8392  
3fff2290:  3fff0fd0 3fff1294 3fff1294 40207e27  
3fff22a0:  3fff1270 00000697 00000697 4010020c  
3fff22b0:  00000001 3fff23a4 3fff6174 4010068c  
3fff22c0:  3fff3a84 3fff2320 3fff2320 3ffe8392  
3fff22d0:  00000000 00000029 3fff0fd8 401004d8  
3fff22e0:  3fff1270 3fff23a4 3fff6174 40212f68  
3fff22f0:  00000001 3fff23a4 3fff2320 4021128c  
3fff2300:  3ffe839c 3fff6174 00000000 3ffe8392  
3fff2310:  00000000 3fff23a4 3fff0fd8 402081a9  
3fff2320:  3ffe9fe0 00000000 000003e8 0000bfd6  
3fff2330:  00000000 00000000 00000000 00000000  
3fff2340:  00000000 feefeffe feefeffe feefeffe  
3fff2350:  feefeffe feefeffe feefeffe 3fff1350  
3fff2360:  3fffdad0 00000000 3fff1349 40213088  
3fff2370:  feefeffe feefeffe 3fff1360 40100720  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vc9f0c112
~ld

@rahu2581 After running your program this is the error message I receive.

rahu2581, 2016-07-08:
@tejaswigowda @musalih Oddly enough I had a similiar crash once, but it works otherwise.
I could not reproduce it locally.

However I think I found the problem: The string length was too short. I committed a potential fix. Please try if this solves the problem.

PR #10 (2016-07-27): Added port number to signed URL
arduinoWebSockets added the port to their Host header.
Since the signature signs this Host header it needs to contain this port also.

Fixes #9. Potentially also some of the other issues.
closed 2016-07-27 by odelot