odelot/aws-mqtt-websockets #11: If internet becomes unavailable; ESP goes in wdt reset

devansh17, 2016-07-09:
Hello,
I checked out the code and works fine with AWS. However, I came across one issue.
After MQTT is subscribed and running, and then at any point after this if the ESP does not get access to internet (Wifi router is on), ESP goes in wdt reset mode. Maybe someone else also encountered this issue and has a fix for it?

thehellmaker, 2016-12-14:
I have the same issue. What I figured out is 
client = new MQTT::Client<IPStack, Countdown, maxMQTTpackageSize, maxMQTTMessageHandlers>(ipstack);

Reduces the memory from 30KB to 10KB and it crashed because of out of memory. Still trying to figure out where the memory is getting consumed.

odelot, 2018-04-15:
@devansh17 it was a bug from arduino/esp8266 sdk  https://github.com/esp8266/Arduino/issues/3537

It is already fixed in the version 2.4.1. 

@thehellmaker I've uploaded working examples with a not dynamic allocated client. You can also use another mqtt client, as pubsubclient. Another way to get memory is reduce the circular buffer that make the interface between the mqtt client and the websocket layer. Remember that this can limit the size of messages sent and received.

`AWSWebSocketClient awsWSclient(1000);  //it is using 1000 bytes`

closed 2018-04-15 by odelot