odelot/aws-mqtt-websockets #12: "please start sntp first !" message while connecting to AWS

hetelek, 2016-07-13:
Hi,

I am trying to get my NodeMCU to publish to AWS IoT. While trying to connect, I get the message:

> please start sntp first !

Multiple times before it finally prints:

> error connection to the web socket server

I am successfully connected to my wifi network. Below is the (semi) full log.

> um 0x2
> m 0x2d
> vc9f0c112
> �connecting to wifi
> state: 5 -> 0 (0)
> rm 0
> f r0, .....................scandone
> .f r0, scan done
> state: 0 -> 2 (b0)
> state: 2 -> 3 (0)
> state: 3 -> 5 (10)
> add 0
> aid 6
> int 
> 
> connected with A Network, channel 7
> dhcp client start...
> ip:192.168.1.30,mask:255.255.255.0,gw:192.168.1.1
> 
> connected
> 31604 - conn: 1 - (34784)
> please start sntp first !
> please start sntp first !
> please start sntp first !
> please start sntp first !
> error connection to the web socket server
> ...

Thank you for any help. I appreciate it.

devansh17, 2016-07-15:
You can ignore this message and refer to other issue, which mentions which Websocket library function you have to use. The latest one doesnt work

thehellmaker, 2016-07-31:
I have the same issue as well. Could you please tell me more about the websocket library function which works?

phlemoine, 2016-08-14:
getting same "sntp" errors. probably worse as it 403 several times before connecting, but then the device freezes and watchdog sends it back to bootloader. log shows a loss of about 18k of memory between conn 1 and conn2. i am using HEAD of each repo listed as dependency (as of aug-12)... i am not sure why commits you guys are using for each of the dependencies - but that might be a valuable info to help reproduce your setups....
in the meantime , i am switching to this solution https://github.com/SuperHouse/esp-open-rtos/pull/173 that allows using certificates. i am not sure how memory footprint will do for my application ... good luck and keep up the good work !

odelot, 2016-11-24:
these messages are debug info from the espwifi library. These are not causing the "error connection to the web socket server". They just appears because of  we turn debug info on in the setup "Serial.setDebugOutput(1);". If you want to see the websocket library debug messages, you need to uncomment this line in the WebSockets.h file. These debug messages will help you find the problem

//#define DEBUG_WEBSOCKETS(...) os_printf( __VA_ARGS__ )

closed 2016-11-24 by odelot