odelot/aws-mqtt-websockets #13: Connected to wss but MQTT connect fail

kami90, 2016-09-03:
I can connect to wss, but when mqtt is connecting, I receive msg "WS-Client: connection lost".
What is my problem?

connected with Tenda_351528, channel 7<\n>
dhcp client start...<\n>
ip:192.168.2.106,mask:255.255.255.0,gw:192.168.2.1<\n>
<\n>
connected<\r><\n>
10746 - conn: 1 - (34336)<\r><\n>
Sat Sep 03 16:40:02 2016<\n>
<\n>
[WS-Client] connect wss...<\n>
[WS-Client] connected to a2o4q1d0iu8ppm.iot.ap-southeast-1.amazonaws.com:443.<\n>
[WS-Client][sendHeader] sending header...<\n>
[WS-Client][sendHeader] sending header... Done (115986us).<\n>
[WS-Client][handleHeader] RX: HTTP/1.1 101 Switching Protocols<\n>
[WS-Client][handleHeader] RX: content-length: 0<\n>
[WS-Client][handleHeader] RX: upgrade: websocket<\n>
[WS-Client][handleHeader] RX: connection: Upgrade<\n>
[WS-Client][handleHeader] RX: sec-websocket-accept: 6rJ86EqKcY0vFEPwKlNgjfzTDXM=<\n>
[WS-Client][handleHeader] RX: sec-websocket-protocol: mqtt<\n>
[WS-Client][handleHeader] RX: access-control-allow-origin: *<\n>
[WS-Client][handleHeader] RX: access-control-allow-headers: Authorization<\n>
[WS-Client][handleHeader] RX: access-control-allow-headers: X-amz-security-token<\n>
[WS-Client][handleHeader] RX: access-control-allow-headers: Accept<\n>
[WS-Client][handleHeader] RX: access-control-allow-headers: X-amz-date<\n>
[WS-Client][handleHeader] RX: access-control-allow-headers: X-amz-user-agent<\n>
[WS-Client][handleHeader] RX: access-control-allow-headers: X-amz-content-sha256<\n>
[WS-Client][handleHeader] RX: access-control-allow-headers: Accept-Language<\n>
[WS-Client][handleHeader] RX: access-control-allow-headers: X-requested-with<\n>
[WS-Client][handleHeader] RX: access-control-allow-headers: Content-Language<\n>
[WS-Client][handleHeader] RX: access-control-allow-headers: Content-Type<\n>
[WS-Client][handleHeader] RX: access-control-expose-headers: x-amzn-ErrorMessage<\n>
[WS-Client][handleHeader] RX: access-control-expose-headers: x-amzn-RequestId<\n>
[WS-Client][handleHeader] RX: access-control-expose-headers: x-amzn-ErrorType<\n>
[WS-Client][handleHeader] RX: access-control-expose-headers: Date<\n>
[WS-Client][handleHeader] Header read fin.<\n>
[WS-Client][handleHeader] Client settings:<\n>
[WS-Client][handleHeader]  - cURL: /mqtt?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIA4OLXG2IIQVMKHQ%2F20160903%2Fap-southeast-1%2Fiotdevicegateway%2Faws4_request&X-Amz-Date=20160903T094001Z&X-Amz-Expires=86400&X-Amz-SignedHeaders=host&X-Amz-Signature=49a71885182198397765030d4f85eaf64389527a0f87579337f1d972faeb815a<\n>
[WS-Client][handleHeader]  - cKey: BCXisznV8O6L30IzWI5Y3Q==<\n>
[WS-Client][handleHeader] Server header:<\n>
[WS-Client][handleHeader]  - cCode: 101<\n>
[WS-Client][handleHeader]  - cIsUpgrade: 1<\n>
[WS-Client][handleHeader]  - cIsWebsocket: 1<\n>
[WS-Client][handleHeader]  - cAccept: 6rJ86EqKcY0vFEPwKlNgjfzTDXM=<\n>
[WS-Client][handleHeader]  - cProtocol: mqtt<\n>
[WS-Client][handleHeader]  - cExtensions: <\n>
[WS-Client][handleHeader]  - cVersion: 0<\n>
[WS-Client][handleHeader] Websocket connection init done.<\n>
[WS][0][headerDone] Header Handling Done (0us).<\n>
websocket layer connected<\r><\n>
MQTT connecting<\r><\n>
[MQTT client]creat connect msg done<\r><\n>
[WS][0][sendFrame] ------- send massage frame -------<\n>
[WS][0][sendFrame] fin: 1 opCode: 2 mask: 1 length: 38 headerToPayload: 0<\n>
[WS][0][sendFrame] pack to one TCP package...<\n>
[WS][0][sendFrame] sending Frame Done (101571us).<\n>
[MQTT client]send packet done<\r><\n>
[WS][0][handleWebsocketWaitFor] size: 2 cWsRXsize: 0<\n>
[readCb] n: 2 t: 14735<\n>
[WS][0][handleWebsocketWaitFor][readCb] size: 2 ok: 1<\n>
[WS-Client] connection lost.<\n>
[WS-Client] client disconnected.<\n>
[WS-Client] connect wss...<\n>
[WS-Client] connected to a2o4q1d0iu8ppm.iot.ap-southeast-1.amazonaws.com:443.<\n>
[WS-Client][sendHeader] sending header...<\n>
[WS-Client][sendHeader] sending header... Done (114428us).<\n>
pm open,type:2 0<\n>

copercini, 2016-09-05:
What arduino IDE and ESP core are you using?

kami90, 2016-09-08:
i have used arduino 1.6.8 version and esp8266 library 2.3.0

tommakov, 2017-03-22:
I appear to be having this same problem.  Any solution?  ( I'm using arduino 1.6.12, but don't know where to check the "ESP core". )  Any help would greatly be appreciated!

tommakov, 2017-03-22:
I'm also using "esp8266 library 2.3.0" and the "Dependancies" from a few days ago.

knownothing8, 2017-11-21:
I have same problem. anyone know how to solve issue yet?

knownothing8, 2017-11-21:
Look like it is AWS IAM policy needs to be set accordingly for user or group to have proper access permission right: 

Go to https://console.aws.amazon.com/ 
Then click IAM
Then click policy.  find your policy or create a new policy   
set service to IOT 
set action to iot:*
set resouce to all resources (I have difficulty to know how to set specific resource. So I just set it all for the experiment)

After, I was able to get MQTT connected!!!

jill6, 2018-04-13:
I signed up to thank you knownothing8!
As soon as I changed the policy in IAM, I am able to connect MQTT!
You saved my time, hwah, Great Respect.

odelot, 2018-04-15:
@jill6 @knownothing8  updated readme with this step-by-step procedure to grant iot permission for the user. thank you

closed 2018-04-15 by odelot