odelot/aws-mqtt-websockets #14: Client disconnect - RX: HTTP/1.1 426 Upgrade Required

2dom, 2016-09-21:
Hi,

since this morning I am unable to connecto the Websocket server. I looks like Amazon changed something server side (protocol requirements).

Any thoughts on this?

connected with ABC, channel 6
dhcp client start...
ip:10.11.10.201,mask:255.255.0.0,gw:10.11.0.1
ping 5, timeout 0, total payload 160 bytes, 5013 ms
Free heap size before AWS: 26312
pm open,type:2 0
[WS-Client] connect wss...
please start sntp first !
[WS-Client] connected to 123.iot.us-east-1.amazonaws.com:443.
[WS-Client][sendHeader] sending header...
[WS-Client][sendHeader] sending header... Done (160746us).
[WS-Client][handleHeader] RX: HTTP/1.1 101 Switching Protocols
[WS-Client][handleHeader] RX: content-length: 0
[WS-Client][handleHeader] RX: upgrade: websocket
[WS-Client][handleHeader] RX: connection: upgrade
[WS-Client][handleHeader] RX: sec-websocket-accept: Q0qdgdfdfg1X4EwVA/+NoP4=
[WS-Client][handleHeader] RX: sec-websocket-protocol: mqtt
[WS-Client][handleHeader] Header read fin.
[WS-Client][handleHeader] Client settings:
[WS-Client][handleHeader]  - cURL: /mqtt?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=x&X-Amz-Date=20160921T084332Z&X-Amz-Expires=86400&X-Amz-SignedHeaders=host&X-Amz-Signature=54645645645645
[WS-Client][handleHeader]  - cKey: sdfsfssdf==
[WS-Client][handleHeader] Server header:
[WS-Client][handleHeader]  - cCode: 101
[WS-Client][handleHeader]  - cIsUpgrade: 0
[WS-Client][handleHeader]  - cIsWebsocket: 1
[WS-Client][handleHeader]  - cAccept: fsdfsf/+NoP4=
[WS-Client][handleHeader]  - cProtocol: mqtt
[WS-Client][handleHeader]  - cExtensions: 
[WS-Client][handleHeader]  - cVersion: 0
[WS-Client][handleHeader] no Websocket connection close.
[WS-Client] client disconnected.
[WS-Client] connect wss...
please start sntp first !
[WS-Client] connected to 123.iot.us-east-1.amazonaws.com:443.
[WS-Client][sendHeader] sending header...
[WS-Client][sendHeader] sending header... Done (138230us).
[WS-Client][handleHeader] RX: HTTP/1.1 426 Upgrade Required
[WS-Client][handleHeader] RX: content-type: application/json
[WS-Client][handleHeader] RX: content-length: 134
[WS-Client][handleHeader] RX: date: Wed, 21 Sep 2016 08:43:36 GMT
[WS-Client][handleHeader] RX: x-amzn-RequestId: 34140c5d-97b9-22fc-89e2-ba79daf799d0
[WS-Client][handleHeader] RX: connection: Keep-Alive
[WS-Client][handleHeader] RX: x-amzn-ErrorType: InternalFailureException:
[WS-Client][handleHeader] RX: sec-websocket-version: 13,8,7
[WS-Client][handleHeader] Header read fin.
[WS-Client][handleHeader] Client settings:
[WS-Client][handleHeader]  - cURL: /mqtt?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=sfsddsfsfsdf%sfsdf%c-east-1%2Fiotdevicegateway%2Faws4_request&X-Amz-Date=20160921T084332Z&X-Amz-Expires=86400&X-Amz-SignedHeaders=host&X-Amz-Signature=sfsdfsdfdds
[WS-Client][handleHeader]  - cKey: sfdsdfsdfd==
[WS-Client][handleHeader] Server header:
[WS-Client][handleHeader]  - cCode: 426
[WS-Client][handleHeader]  - cIsUpgrade: 0
[WS-Client][handleHeader]  - cIsWebsocket: 0
[WS-Client][handleHeader]  - cAccept: 
[WS-Client][handleHeader]  - cProtocol: 
[WS-Client][handleHeader]  - cExtensions: 
[WS-Client][handleHeader]  - cVersion: 13
[WS-Client][handleHeader] no Websocket connection close.
[WS-Client] client disconnected.

samehhady, 2016-09-22:
I have the same issue!

It was working perfectly for 2 days and then suddenly yesterday it stopped working giving me:

```
[WS-Client][sendHeader] sending header...
[WS-Client][sendHeader] sending header... Done (239475us).
[WS-Client][handleHeader] RX: HTTP/1.1 426 Upgrade Required
[WS-Client][handleHeader] RX: content-type: application/json
[WS-Client][handleHeader] RX: content-length: 134
[WS-Client][handleHeader] RX: date: Thu, 22 Sep 2016 06:00:39 GMT
[WS-Client][handleHeader] RX: x-amzn-RequestId: 54a7690c-00bc-fa4a-d822-26643daafae4
[WS-Client][handleHeader] RX: connection: Keep-Alive
[WS-Client][handleHeader] RX: x-amzn-ErrorType: InternalFailureException:
[WS-Client][handleHeader] RX: sec-websocket-version: 13,8,7
[WS-Client][handleHeader] Header read fin.
[WS-Client][handleHeader] Client settings:
[WS-Client][handleHeader]  - cURL: /mqtt?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAJWOFIGFI46YPZLRA%2F20160922%2Fus-east-1%2Fiotdevicegateway%2Faws4_request&X-Amz-Date=20160922T060033Z&X-Amz-Expires=86400&X-Amz-SignedHeaders=host&X-Amz-Signature=74fda29e770309d5f0b6fcc1e54105e0b9f170e928919d5c438bc6ec78b21c21
[WS-Client][handleHeader]  - cKey: c2ZQnlS2XwdsGxMj2hXF6g==
[WS-Client][handleHeader] Server header:
[WS-Client][handleHeader]  - cCode: 426
[WS-Client][handleHeader]  - cIsUpgrade: 0
[WS-Client][handleHeader]  - cIsWebsocket: 0
[WS-Client][handleHeader]  - cAccept: 
[WS-Client][handleHeader]  - cProtocol: 
[WS-Client][handleHeader]  - cExtensions: 
[WS-Client][handleHeader]  - cVersion: 13
[WS-Client][handleHeader] no Websocket connection close.
[WS-Client] client disconnected.
[AWSc] Disconnected!
error connection to the websocket server
```

I didn't change anything yesterday, actually I was going to keep it up and running for a week to check how reliable the connection is.

100% sure it is related to amazon changing something in their end.

samehhady, 2016-09-22:
Maybe that would help, after some searching I found the reason of that response from AWS

based on: http://tools.ietf.org/html/rfc6455#section-4.2.2

> The |Sec-WebSocket-Version| header field in the client's
>       handshake includes the version of the WebSocket Protocol with
>       which the client is attempting to communicate.  If this
>       version does not match a version understood by the server, the
>       server MUST abort the WebSocket handshake described in this
>       section and instead send an appropriate HTTP error code (such
>       as 426 Upgrade Required) and a |Sec-WebSocket-Version| header
>       field indicating the version(s) the server is capable of
>       understanding.

I am still trying to understand why the Sec-WebSocket-Version is wrong, I can see we are sending Sec-WebSocket-Version: 13 which should be acceptable by AWS since they accept 13, 8, 7

boraozgen, 2016-09-22:
Hi, I have the same issue too. I am going to contact the AWS support over this. Any ideas would be appreciated.

samehhady, 2016-09-22:
very possible, I've seen discussions that TLS 1.2 is being implemented now in RTOS version on ESP, lets hope igrr implement it soon on Arduino version.

Still if that's the case then every single ESP using this library should not work with him and we should find lots of replies already!

samehhady, 2016-09-22:
OK, I tracked down the issue and finally fixed it!

open WebSocketClient.cpp and search for:

`if(headerValue.indexOf("Upgrade") >= 0) {`

replace it with: 

`if(headerValue.equalsIgnoreCase("Upgrade") >= 0) {`

And it should work, as simple as that!

I guess both of us are using the older version of the websocket client :D

boraozgen, 2016-09-23:
Wow, nice find! That worked out for me too. But I suppose we have to find a backup way if things get messed up AWS side like it happened this week.

BTW, I tried to update my aws-mqtt-websockets and arduinoWebsockets libraries in hope for a fix yesterday, however I kept getting the 403 error as mentioned in other issue topics. Is there still no solution for that, other than using an older version of arduinoWebsockets?

2dom, 2016-09-23:
Hey ... thanks ... that did the trick. I am wondering how many devices they disconnected by changing that string...

samehhady, 2016-09-23:
Tomorrow i will look more into the new version and will try to make it work, most likely different header is sent and thats why AWS refuse it.

odelot, 2016-11-24:
ArduinoWebsockets library version 2.0.5 already fixed this issue. Now the library works with it without moddification

closed 2016-11-24 by odelot