odelot/aws-mqtt-websockets #15: Problem After Publish 6 ou 7 Times

vitoralvimb, 2016-11-09:
Hello Odelot / Other Friends, 

I am trying to use your solution, I can connect with AWS IOT and publish something, but I found a problem and I could not fix yet. Could you help me, please?

After publish 6 or 7 times, I get the error below. I've tried many ways to resolve this problem, but it is very hard!

I just put this code at your example:

`if (connect ()){
      subscribe ();
      sendmessage(); delay(5000);
      sendmessage(); delay(5000);
      sendmessage(); delay(5000);
      sendmessage(); delay(5000);
      sendmessage(); delay(5000);
      sendmessage(); delay(5000);
      sendmessage(); delay(5000);
      sendmessage(); delay(5000);
      sendmessage(); delay(5000);
      sendmessage(); delay(5000);
      sendmessage(); delay(5000);
      sendmessage(); delay(5000);
      sendmessage(); delay(5000);
}`

And you can see the LOG:
`MQTT connected
[WS][0][sendFrame] ------- send massage frame -------
[WS][0][sendFrame] fin: 1 opCode: 2 mask: 1 length: 57 headerToPayload: 0
[WS][0][sendFrame] pack to one TCP package...
[WS][0][sendFrame] sending Frame Done (1758us).
[WS][0][handleWebsocketWaitFor] size: 2 cWsRXsize: 0
[readCb] n: 2 t: 7283
[WS][0][handleWebsocketWaitFor][readCb] size: 2 ok: 1
[WS][0][handleWebsocket] ------- read massage frame -------
[WS][0][handleWebsocket] fin: 1 rsv1: 0 rsv2: 0 rsv3 0  opCode: 2
[WS][0][handleWebsocket] mask: 0 payloadLen: 5
[readCb] n: 5 t: 7298
[WSc] get binary length: 5
MQTT subscribed
Publishing... {"state":{"reported":{"on": false}, "desired":{"on": false}}}
[WS][0][sendFrame] ------- send massage frame -------
[WS][0][sendFrame] fin: 1 opCode: 2 mask: 1 length: 116 headerToPayload: 0
[WS][0][sendFrame] pack to one TCP package...
[WS][0][sendFrame] sending Frame Done (1902us).
Publishing... {"state":{"reported":{"on": false}, "desired":{"on": false}}}
[WS][0][sendFrame] ------- send massage frame -------
[WS][0][sendFrame] fin: 1 opCode: 2 mask: 1 length: 116 headerToPayload: 0
[WS][0][sendFrame] pack to one TCP package...
[WS][0][sendFrame] sending Frame Done (1944us).
Publishing... {"state":{"reported":{"on": false}, "desired":{"on": false}}}
[WS][0][sendFrame] ------- send massage frame -------
[WS][0][sendFrame] fin: 1 opCode: 2 mask: 1 length: 116 headerToPayload: 0
[WS][0][sendFrame] pack to one TCP package...
[WS][0][sendFrame] sending Frame Done (1946us).
Publishing... {"state":{"reported":{"on": false}, "desired":{"on": false}}}
[WS][0][sendFrame] ------- send massage frame -------
[WS][0][sendFrame] fin: 1 opCode: 2 mask: 1 length: 116 headerToPayload: 0
[WS][0][sendFrame] pack to one TCP package...
[WS][0][sendFrame] sending Frame Done (1960us).
Publishing... {"state":{"reported":{"on": false}, "desired":{"on": false}}}
[WS][0][sendFrame] ------- send massage frame -------
[WS][0][sendFrame] fin: 1 opCode: 2 mask: 1 length: 116 headerToPayload: 0
[WS][0][sendFrame] pack to one TCP package...
[WS][0][sendFrame] sending Frame Done (2048us).
Publishing... {"state":{"reported":{"on": false}, "desired":{"on": false}}}
[WS][0][sendFrame] ------- send massage frame -------
[WS][0][sendFrame] fin: 1 opCode: 2 mask: 1 length: 116 headerToPayload: 0
[WS][0][sendFrame] pack to one TCP package...
[WS][0][sendFrame] sending Frame Done (1975us).
...
...
...
...
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
Publishing... {"state":{"reported":{"on": false}, "desired":{"on": false}}}
[WS][0][sendFrame] ------- send massage frame -------
[WS][0][sendFrame] fin: 1 opCode: 2 mask: 1 length: 116 headerToPayload: 0

Soft WDT reset

ctx: cont 
sp: 3fff27a0 end: 3fff2c70 offset: 01b0

>>>stack>>>
3fff2950:  3fff1b64 00000309 00000309 4010020c  
3fff2960:  3fff5ba4 00000035 3fff51cc 4010068c  
3fff2970:  00000035 cafbd116 00000000 40209ea8  
3fff2980:  0000000a 00000009 3fff7870 4020424b  
3fff2990:  3fff5ba4 00000035 3fff51cc 40204728  
3fff29a0:  3fff77c0 3fff77d0 3fff79a0 00000000  
3fff29b0:  3fff5ba4 00000035 3fff51cc 402052b0  
3fff29c0:  00000000 3fff5aec 00000035 402367c8  
3fff29d0:  00020317 3fff1806 00000040 01000000  
3fff29e0:  00000006 00000017 00000010 3fff51ac  
3fff29f0:  00000000 3fff1838 40209f70 00000000  
3fff2a00:  3fff5aec 3fff2a60 00000006 4023684f  
3fff2a10:  00000006 40206a48 00000074 40202974  
3fff2a20:  00000000 00000b29 00000076 3fff1838  
3fff2a30:  00000000 3fff521c 3fff51ac 40204d2c  
3fff2a40:  00000001 00000074 00000074 40206c0a  
3fff2a50:  00000000 3fff521c 40209f70 3fff1c50  
3fff2a60:  0000f482 00000000 00000000 00000000  
3fff2a70:  3fff0000 3fff51ac 3fff1c50 4020a028  
3fff2a80:  00000006 03b96e08 00000000 00000002  
3fff2a90:  00000001 3fff8ee6 00000000 40204f7b  
3fff2aa0:  40104ec4 3fff1818 3fff1838 3fff521c  
3fff2ab0:  00000074 3fff1838 3fff1818 402077e0  
3fff2ac0:  00000001 00000000 00000002 4020947b  
3fff2ad0:  00000000 3fff1804 3fff1808 0000000a  
3fff2ae0:  00000000 3fff1804 00000003 3fff2c10  
3fff2af0:  00000074 3fff5214 00000074 40207800  
3fff2b00:  0000382b 00000000 00000000 402082d4  
3fff2b10:  00000072 3fff5214 00000000 40202d0d  
3fff2b20:  3fff1804 0001693d 3fff521c 00000074  
3fff2b30:  3fffdad0 3fff521c 00000000 00000000  
3fff2b40:  3fff2c10 3fff5214 3fff2ba0 402030a8  
3fff2b50:  3fff521d 00000000 00000000 40209294  
3fff2b60:  00000003 00000001 3fff2c10 00000001  
3fff2b70:  00000074 3fff5214 3fff2ba0 402031d0  
3fff2b80:  3ffe96cc 00000000 00000000 3fff2ba0  
3fff2b90:  0000003e 00000001 3fff561c 40201831  
3fff2ba0:  7473227b 22657461 72227b3a 726f7065  
3fff2bb0:  22646574 6f227b3a 203a226e 736c6166  
3fff2bc0:  202c7d65 73656422 64657269 227b3a22  
3fff2bd0:  3a226e6f 6c616620 7d7d6573 4020007d  
3fff2be0:  3fff2c10 0000000e 3ffe968a 3fff1c44  
3fff2bf0:  3fffdad0 0000000f 3fff1b88 40208171  
3fff2c00:  40104ec4 3ffe96cc 00000000 00000000  
3fff2c10:  0001693d 3fff19d8 3fff1c50 3fff19d8  
3fff2c20:  3fff521c 3fff1c44 40209f70 3fff1c50  
3fff2c30:  40201b7e 00001388 00001388 3fff1c44  
3fff2c40:  3fffdad0 3fff1938 00001388 402036cd  
3fff2c50:  feefeffe 00000000 3fff1c3d 40209fb4  
3fff2c60:  feefeffe feefeffe 3fff1c50 40100720  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)`

vitoralvimb, 2016-11-24:
@odelot (Fabio), I was using the MQTT Paho from PlatformIO/Arduino. This is an old version and I think it is the problem.
I changed this library by the newest one from https://projects.eclipse.org/projects/technology.paho/downloads

Now it is working OK!!!

odelot, 2016-11-24:
\o/

closed 2016-11-24 by odelot