odelot/aws-mqtt-websockets #18: aws-iot with esp8266 error

ashwnaipal, 2017-02-07:
please help me on connecting with aws iot services with esp8266 12e. i am using example aws-mqtt-websocket example not able to connect with dashboard of aws iot the code is as follow and the error i get is (please start sntp first ! error connection to the websocket server)

`#include <Arduino.h>
  #include <Stream.h>

  #include <ESP8266WiFi.h>

  //AWS
  #include "sha256.h"
  #include "Utils.h"
  #include "AWSClient2.h"

  //WEBSockets
  #include <Hash.h>
  #include <WebSocketsClient.h>

  //MQTT PAHO
  #include <SPI.h>
  #include <IPStack.h>
  #include <Countdown.h>
  #include <MQTTClient.h>

  //AWS MQTT Websocket
  #include "Client.h"
  #include "AWSWebSocketClient.h"
  #include "CircularByteBuffer.h"

  //AWS IOT config, change these:

  const char* ssid = "NPSE";
  const char* password = "P/W070415";
  char aws_endpoint[]    = "aph0r6gk1etz5.iot.us-east-1.amazonaws.com";
  char aws_key[]         = "AKIAJDJA6HPCGZXG7N6Q";
  char aws_secret[]      = "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";
  char aws_region[]      = "us-east-1";
  const char* aws_topic  = "$aws/things/tutorial/shadow/update";
  int port = 443;

  //MQTT config
  const int maxMQTTpackageSize = 512;
  const int maxMQTTMessageHandlers = 1;

  WiFiClientSecure espclient;

  AWSWebSocketClient awsWSclient(1000);

  IPStack ipstack(awsWSclient);

  MQTT::Client<IPStack, Countdown, maxMQTTpackageSize, maxMQTTMessageHandlers> *client = NULL;

  //# of connections
  long connection = 0;

  //generate random mqtt clientID
  char* generateClientID () {
    char* cID = new char[23]();
    for (int i=0; i<22; i+=1)
      cID[i]=(char)random(1, 256);
    return cID;
  }

  //count messages arrived
  int arrivedcount = 0;

  //callback to handle mqtt messages
  void messageArrived(MQTT::MessageData& md)
  {
    MQTT::Message &message = md.message;

    Serial.print("Message ");
    Serial.print(++arrivedcount);
    Serial.print(" arrived: qos ");
    Serial.print(message.qos);
    Serial.print(", retained ");
    Serial.print(message.retained);
    Serial.print(", dup ");
    Serial.print(message.dup);
    Serial.print(", packetid ");
    Serial.println(message.id);
    Serial.print("Payload ");
    char* msg = new char[message.payloadlen+1]();
    memcpy (msg,message.payload,message.payloadlen);
    Serial.println(msg);
    delete msg;
  }

  //connects to websocket layer and mqtt layer
  bool connect () {

      if (client == NULL) {
        client = new MQTT::Client<IPStack, Countdown, maxMQTTpackageSize, maxMQTTMessageHandlers>(ipstack);
      } else {

        if (client->isConnected ()) {    
          client->disconnect ();
        }  
        delete client;
        client = new MQTT::Client<IPStack, Countdown, maxMQTTpackageSize, maxMQTTMessageHandlers>(ipstack);
      }

      //delay is not necessary... it just help us to get a "trustful" heap space value
      delay (1000);
      Serial.print (millis ());
      Serial.print (" - conn: ");
      Serial.print (++connection);
      Serial.print (" - (");
      Serial.print (ESP.getFreeHeap ());
      Serial.println (")");

     int rc = ipstack.connect(aws_endpoint, port);
     Serial.print(rc);

      if (rc != 1)
      {
        Serial.println("error connection to the websocket server");
        return false;
      } else {
        Serial.println("websocket layer connected");
      }

      Serial.println("MQTT connecting");
      MQTTPacket_connectData data = MQTTPacket_connectData_initializer;
      data.MQTTVersion = 3;
      char* clientID = generateClientID ();
      data.clientID.cstring = clientID;
      rc = client->connect(data);
      delete[] clientID;
      if (rc != 0)
      {
        Serial.print("error connection to MQTT server");
        Serial.println(rc);
        return false;
      }
      Serial.println("MQTT connected");
      return true;
  }

  //subscribe to a mqtt topic
  void subscribe () {
     //subscript to a topic
      int rc = client->subscribe(aws_topic, MQTT::QOS0, messageArrived);
      Serial.println(rc);
      if (rc != 0) {
        Serial.print("rc from MQTT subscribe is ");
        Serial.println(rc);
        return;
      }
      Serial.println("MQTT subscribed");
  }

  //send a message to a mqtt topic
  void sendmessage () {
      //send a message
      MQTT::Message message;
      char buf[100];
      strcpy(buf, "{\"state\":{\"reported\":{\"on\": false}, \"desired\":{\"on\": false}}}");
      message.qos = MQTT::QOS0;
      message.retained = false;
      message.dup = false;
      message.payload = (void*)buf;
      message.payloadlen = strlen(buf)+1;
      int rc = client->publish(aws_topic, message); 
  }

  void setup_wifi() {

    delay(10);
    // We start by connecting to a WiFi network
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(ssid);

    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
  }

  void setup() {
      Serial.begin (115200);
      setup_wifi();

      Serial.setDebugOutput(1); 
     //fill AWS parameters    
      awsWSclient.setAWSRegion(aws_region);
      awsWSclient.setAWSDomain(aws_endpoint);
      awsWSclient.setAWSKeyID(aws_key);
      awsWSclient.setAWSSecretKey(aws_secret);
      awsWSclient.setUseSSL(true);

      if (connect ()){
        subscribe ();
        sendmessage ();
      }

  }

  void loop() {
    //keep the mqtt up and running
    if (awsWSclient.connected ()) {    
        client->yield();
    } else {
      //handle reconnection
      if (connect ()){
        subscribe ();      
      }
    }

  }`
[image]

sh4nnongoh, 2017-02-15:
I am having the same issue: "Please start SNTP first"

moodgorning, 2017-02-26:
put `configTime(3 * 3600, 0, "pool.ntp.org", "time.nist.gov");` in your setup to get rid of the Please start sntp first message, It is however not going to solve your problem. The sntp message can be ignored

closed 2017-06-02 by odelot