odelot/aws-mqtt-websockets #20: Need subscribe example

rileyschuit, 2017-02-20:
I have trying to figure out how to get the desired shadow from AWS.  Publishing the state is no issue.

I think I found the solution here:  https://github.com/heskew/aws-sdk-arduino/blob/iot-get-shadow-and-cleanup/src/common/AmazonIOTClient.cpp but I'm failing to put it together.  I have attempted to reach out to the other without success.

If we can get the subscribe function in the example (https://github.com/odelot/aws-mqtt-websockets/blob/master/examples/aws-mqtt-websocket-example/aws-mqtt-websocket-example.ino)  to return the value from AWS, we have a full solution.

I have reached out to you via twitter to offer compensation for your efforts as well.

Huong-nt, 2017-02-21:
You can get the desired shadow from AWS by publishing to the topic $aws/things/myThingName/shadow/get and subscribe topic $aws/things/myThingName/shadow/get/accepted.
When you publish a message to the topic $aws/things/myThingName/shadow/get, AWS IoT will publish a response state document to $aws/things/myThingName/shadow/get/accepted.

rileyschuit, 2017-02-22:
@Huong-nt This worked!  Thank you so much!  I will submit another pull request with an example when it functions correctly and is somewhat clean.  With the example given, I modded/added:
```
const char* aws_topic_publish  = "$aws/things/Device1/shadow/get";
const char* aws_topic_subscribe  = "$aws/things/Device1/shadow/get/accepted";
```

Then modified the subscribe function like so:
```
//subscribe to a mqtt topic
void subscribe () {
   //subscrip to a topic
    int rc = client->subscribe(aws_topic_subscribe, MQTT::QOS0, messageArrived);
    if (rc != 0) {
      if (DEBUG_PRINT) {
        Serial.print("rc from MQTT subscribe is ");
        Serial.println(rc);
      }
      return;
    }
    if (DEBUG_PRINT) {
      Serial.println("MQTT subscribed");
    }
}
```

Serial output looks like:

```
MQTT subscribed
Message 1 arrived: qos 0, retained 0, dup 0, packetid 16383
Payload {"state":{"desired":{"on":false},"reported":{"on":false}},"metadata":{"desired":{"on":{"timestamp":1487734811}},"reported":{"on":{"timestamp":1487734811}}},"version":36,"timestamp":1487735007}

```

closed 2017-02-22 by rileyschuit