odelot/aws-mqtt-websockets #21: mqtt-websockets for mosquitto

zenonasz, 2017-02-21:
Hi,

Any chance of porting the library so it can support the mosquitto broker?

Havent try to use it as it is with the mosquitto broker but as i understand the technology is similar.

Thanks,

Zen

odelot, 2017-06-02:
probably the mosquito broker secures the connection via https, so you don't need to use MQTT over websockets as we need to use with AWS

You can use a MQTT library directly, like the Paho MQTT library I am using in this solution.

zenonasz, 2017-06-27:
of course you need to. 
if you want to reach mqtt through 443 or port 80 and not punch a hole in your router.

odelot, 2017-06-27:
The idea of this library is provide a way to use **AWS** MQTT with ESP8266. The use of the websockets protocol is needed because when the library was created, the esp cannot handle tls 1.2 and thus cannot connect to the MQTT server.

when I was building the library, I designed it to override the AWS signing step to use mqtt over websockets (they use their sv4 sign procedure together with HTTPs.

I never tested it, but here is how to use it:

Instead of configure the library with AWS stuff like this:

```
awsWSclient.setAWSRegion(aws_region);
awsWSclient.setAWSDomain(aws_endpoint);
awsWSclient.setAWSKeyID(aws_key);
awsWSclient.setAWSSecretKey(aws_secret);
```

you use just:

`awsWSclient.setPath ("mypahomqttwebsocketpath.com");`

if you want to use https, please add:

`awsWSclient.setUseSSL(true);`

If it works, it would be nice to publish as an example of use :-)

AND PLEASE DO NOT PUNCH A HOLE IN YOUR ROUTER! use the firewall ha ha ha

Vayatoalla, 2017-12-27:
Well. I tried this with a Mosquitto server and it works.
I am using SSL.
Only thing that I cannot resolve yet is how to configure the connection with user and password.

Regards!