odelot/aws-mqtt-websockets #23: error: conflicting return type specified for 'virtual void Sha256Class::write(uint8_t)'

zaifz, 2017-02-28:
[image]

I'm using Generic ESP8266 module and trying to establish a secure SSL connection.
I get the following error after installing all the libraries:

`
In file included from /Users/huzaifah/Downloads/aws-mqtt-websockets-master/examples/aws-mqtt-websocket-example/aws-mqtt-websocket-example.ino:8:0:
/Users/huzaifah/Documents/Arduino/libraries/Others/sha256.h:26:18: error: conflicting return type specified for 'virtual void Sha256Class::write(uint8_t)'
     virtual void write(uint8_t);
                  ^
In file included from /Users/huzaifah/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/Stream.h:26:0,
                 from /Users/huzaifah/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/HardwareSerial.h:31,
                 from /Users/huzaifah/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/Arduino.h:245,
                 from /Users/huzaifah/Downloads/aws-mqtt-websockets-master/examples/aws-mqtt-websocket-example/aws-mqtt-websocket-example.ino:1:
/Users/huzaifah/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/Print.h:55:24: error:   overriding 'virtual size_t Print::write(uint8_t)'
         virtual size_t write(uint8_t) = 0;
                        ^
aws-mqtt-websocket-example:48: error: 'MQTT' does not name a type
 MQTT::Client<IPStack, Countdown, maxMQTTpackageSize, maxMQTTMessageHandlers> *client = NULL;
 ^
aws-mqtt-websocket-example:65: error: variable or field 'messageArrived' declared void
 void messageArrived(MQTT::MessageData& md)
                     ^
aws-mqtt-websocket-example:65: error: 'MQTT' has not been declared
aws-mqtt-websocket-example:65: error: 'md' was not declared in this scope
 void messageArrived(MQTT::MessageData& md)
                                        ^
aws-mqtt-websocket-example:65: error: variable or field 'messageArrived' declared void
 void messageArrived(MQTT::MessageData& md)
                     ^
aws-mqtt-websocket-example:65: error: 'MQTT' has not been declared
aws-mqtt-websocket-example:65: error: 'md' was not declared in this scope
 void messageArrived(MQTT::MessageData& md)
                                        ^
/Users/huzaifah/Downloads/aws-mqtt-websockets-master/examples/aws-mqtt-websocket-example/aws-mqtt-websocket-example.ino: In function 'bool connect()':
aws-mqtt-websocket-example:89: error: 'client' was not declared in this scope
     if (client == NULL) {
         ^
aws-mqtt-websocket-example:90: error: expected type-specifier before 'MQTT'
       client = new MQTT::Client<IPStack, Countdown, maxMQTTpackageSize, maxMQTTMessageHandlers>(ipstack);
                    ^
aws-mqtt-websocket-example:90: error: expected ';' before 'MQTT'
aws-mqtt-websocket-example:96: error: type '<type error>' argument given to 'delete', expected pointer
       delete client;
              ^
aws-mqtt-websocket-example:97: error: expected type-specifier before 'MQTT'
       client = new MQTT::Client<IPStack, Countdown, maxMQTTpackageSize, maxMQTTMessageHandlers>(ipstack);
                    ^
aws-mqtt-websocket-example:97: error: expected ';' before 'MQTT'
aws-mqtt-websocket-example:128: error: 'client' was not declared in this scope
     rc = client->connect(data);
          ^
/Users/huzaifah/Downloads/aws-mqtt-websockets-master/examples/aws-mqtt-websocket-example/aws-mqtt-websocket-example.ino: In function 'void subscribe()':
aws-mqtt-websocket-example:143: error: 'client' was not declared in this scope
     int rc = client->subscribe(aws_topic, MQTT::QOS0, messageArrived);
              ^
aws-mqtt-websocket-example:143: error: 'MQTT' has not been declared
     int rc = client->subscribe(aws_topic, MQTT::QOS0, messageArrived);
                                           ^
aws-mqtt-websocket-example:143: error: 'messageArrived' was not declared in this scope
     int rc = client->subscribe(aws_topic, MQTT::QOS0, messageArrived);
                                                       ^
/Users/huzaifah/Downloads/aws-mqtt-websockets-master/examples/aws-mqtt-websocket-example/aws-mqtt-websocket-example.ino: In function 'void sendmessage()':
aws-mqtt-websocket-example:155: error: 'MQTT' has not been declared
     MQTT::Message message;
     ^
aws-mqtt-websocket-example:155: error: expected ';' before 'message'
     MQTT::Message message;
                   ^
aws-mqtt-websocket-example:158: error: 'message' was not declared in this scope
     message.qos = MQTT::QOS0;
     ^
aws-mqtt-websocket-example:158: error: 'MQTT' has not been declared
     message.qos = MQTT::QOS0;
                   ^
aws-mqtt-websocket-example:163: error: 'client' was not declared in this scope
     int rc = client->publish(aws_topic, message); 
              ^
/Users/huzaifah/Downloads/aws-mqtt-websockets-master/examples/aws-mqtt-websocket-example/aws-mqtt-websocket-example.ino: In function 'void loop()':
aws-mqtt-websocket-example:198: error: 'client' was not declared in this scope
       client->yield();
       ^
exit status 1
'MQTT' does not name a type`

Kindly help me to analyze the situation and find a solution for the same.

odelot, 2017-06-02:
You need to remove this library first /Users/huzaifah/Documents/Arduino/libraries/Others/sha256.h

zaifz, 2017-06-19:
Thank you.. This worked :)

closed 2017-06-19 by zaifz