odelot/aws-mqtt-websockets #25: How to subscribe to multiple topic using paho MQTT in arduino

sunilsud, 2017-08-04:
i am able to pulish to multiple topic
how do i subscribe to multiple topic1 and topic 2
right now only topic1 i can subscribe

void subscribe(){

   //subscript to a topic
   

 client->subscribe(topic1,MQTT::QOS0,messageArrived1);

 client->subscribe(topic2,MQTT::QOS0,messageArrived2);

}

void messageArrived1(MQTT::MessageData& md)
{
  MQTT::Message &message = md.message;
 
 char* msg = new char[message.payloadlen+1]();
  memcpy (msg,message.payload,message.payloadlen);
 json_msg = msg;
Serial.print(json_msg);
parse_json();
delete msg;

}
void messageArrived2(MQTT::MessageData& Md)
{
  MQTT::Message &message = Md.message;
 
 char* msg = new char[message.payloadlen+1]();
  memcpy (msg,message.payload,message.payloadlen);
 json_msg = msg;
Serial.print(json_msg);
parse_json();
delete msg;

}

odelot, 2017-08-04:
When you create the client, you need to inform the max number os message handlers.

In the example, it is set to 1. Use 2.

```
const int maxMQTTMessageHandlers = 1;
MQTT::Client<IPStack, Countdown, maxMQTTpackageSize, maxMQTTMessageHandlers> *client = NULL;
```

sunilsud, 2017-08-05:
its working

closed 2017-08-05 by sunilsud

rakeshcomavia, 2017-09-07:
thank you for help, its working