odelot/aws-mqtt-websockets #28: MQTT publish big size payload

waghekapil, 2017-10-27:
Hello,

I have big size payload to send through AWS MQTT. Paload size is more than 800 characters. For small size string it is working fine but not with big size.

Can you please share any way to free the `Message` objects or `buf `?

Thanks

waghekapil, 2017-11-06:
[image]

Large amount of Heap memory is consumed by library. Can you please help me into this?
I'm using IR library to receive IR data. Due to low memory it's restarting ESP.

odelot, 2017-11-06:
You can decrease the maxMQTTpackageSize and decrease the size of the circular buffer that is used between the MQTT client and the Websocket Tunnel.

In the example I think it is using 10000 bytes.  You can decrease to a number that can handle at least more than one mqtt message (to handle the scenario that two or more mqtt messages arrived in the websocket tunnel but was not processed by the mqtt client yet)

In a production project I am using this (circular buffer in 5000 bytes):

```
const int maxMQTTpackageSize = 256;
const int maxMQTTMessageHandlers = 1;

//websocket transport layer
AWSWebSocketClient awsWSclient(256,5000);
```

waghekapil, 2017-11-06:
Thanks for the reply.

`//websocket transport layer
AWSWebSocketClient awsWSclient(256,5000);`

Not able to pass two parameters. Constructor has only one parameter to pass.
`//bufferSize defines the size of the circular byte buffer that provides the interface between messages arrived in websocket layer and byte reads from mqtt layer	
  AWSWebSocketClient (unsigned int bufferSize = 1000);`

I have upto 1500 characters to send to the AWS IoT.

waghekapil, 2017-11-06:
My apologies. Just checked the updated version of file.

Let me check it and get back to you.

odelot, 2017-11-06:
Sorry, the 5000 is a timeout in seconds. the first parameter is the circular buffer size. I am using 256 bytes. and decreased the maxMQTTpackageSize.

Update the library. I move almost everything to static memory because I was having problems with memory fragmentation.

waghekapil, 2017-11-06:
Getting error:
 `In file included from F:\Working Folder\Saleh\Arduino\Arduino_updated\Controllex_Updated\Controllex_Updated.ino:37:0:

C:\Users\kapil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\aws-mqtt-websockets-master/AWSWebSocketClient.h:12:18: error: redefinition of 'const int HASH_HEX_LEN2'

 static const int HASH_HEX_LEN2 = 64;

                  ^

In file included from F:\Working Folder\Saleh\Arduino\Arduino_updated\Controllex_Updated\Controllex_Updated.ino:23:0:

C:\Users\kapil\Documents\Arduino\libraries\AWS-SDK-ESP8266\src/AWSClient2.h:21:18: error: 'const int HASH_HEX_LEN2' previously defined here

 static const int HASH_HEX_LEN2 = 64;

                  ^

Multiple libraries were found for "sha256.h"
 Used: C:\Users\kapil\Documents\Arduino\libraries\AWS-SDK-ESP8266
 Not used: C:\Users\kapil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\AWSArduinoSDK
 Not used: C:\Users\kapil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\AWSArduinoSDK
 Not used: C:\Users\kapil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\AWSArduinoSDK
 Not used: C:\Users\kapil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\AWSArduinoSDK
exit status 1
Error compiling for board Generic ESP8266 Module.`

odelot, 2017-11-06:
remove this include AWSClient2.h from your .ino

waghekapil, 2017-11-06:
Thanks. 
Now, I'm not getting any error. It is processing with 1k characters string. But Why I'm not getting it published?

Publish function is not working now.

odelot, 2017-11-06:
If the message you are sending is up to 1500 character, you need to have a bigger circular buffer and maxMQTTpackageSize, something bigger than 3000. 

try sending smaller messages to see if it is working. I am using this library version without problems

waghekapil, 2017-11-06:
Thanks-a-lot :)

You saved me and my job too. I really appreciate your help. I was trying to solve this issue since last 3 weeks. 

I published "Welcome!"message and it's working. So, I think I have to play with maxMQTTpackageSize only. Correct?

My IR rawData size is around 1000+ characters for AC remotes. I want to send them to server and for execution same want to receive from server through MQTT.

I want to receive the same large size string (1k+ characters in MQTT subscribe topics).

Will getback to you, if I need anything else.

My best wishes for you.

**Thanks! again**