odelot/aws-mqtt-websockets #31: Fatal exception 29 when the internet suddenly goes down

handgear, 2018-01-06:
Hello. First of all thank you for this useful library.
I'm testing for internet disconnection situation or situation that AP disappear.
It works fine for sudden absent of AP. (it recovers the connection automatically when AP is back)
But in case of internet disconnection it has a problem.

This is the log after internet disconnection.

[WS][0][sendFrame] ------- send massage frame -------
[WS][0][sendFrame] fin: 1 opCode: 2 mask: 1 length: 2 headerToPayload: 0
[WS][0][sendFrame] pack to one TCP package...
[write] n: 8 t: 71466
:wr
:ww
:wr
:ww
:wr
:ww
:wr
:ww
:wr
Fatal exception 29(StoreProhibitedCause):
epc1=0x4000e1b2, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

Exception (29):
epc1=0x4000e1b2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffda0 end: 3fffffb0 offset: 01a0

\>>>stack>>>
3fffff40:  00000000 00000000 00000300 4010071d  
3fffff50:  3fffdc00 00000000 3fff0dc0 40107638  
3fffff60:  4010579e 01b41273 3fff1d20 402187b2  
3fffff70:  4022a3fc 3fff1cf8 40218c72 00000002  
3fffff80:  4022a422 3fffdab0 40219e6f 3fffdab0  
3fffff90:  00000000 3fffdab0 3fff26ac 40210cb3  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,0)
(log end)

[stack decode result]
Decoding 12 results
0x4010071d: calloc at C:\Users\handgear\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1684
0x40107638: pvPortZalloc at C:\Users\handgear\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/heap.c line 33
0x4010579e: wdt_feed at ?? line ?
0x402187b2: pm_set_sleep_time at ?? line ?
0x4022a3fc: ets_timer_handler_isr at ?? line ?
0x40218c72: pm_get_sleep_type at ?? line ?
0x4022a422: ets_timer_handler_isr at ?? line ?
0x40219e6f: pm_post at ?? line ?
0x40210cb3: loop_task at C:\Users\handgear\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56

i guess arduinoWebSockets could be the problem though 
don't know why it dosn't exit from here ->
[https://github.com/Links2004/arduinoWebSockets/blob/642750e232e3f0dbb23a10c0f12772df0e98b087/src/WebSockets.cpp#L618](url)

Or shoud i ask to Links2004/arduinoWebSockets ?

odelot, 2018-01-07:
Hi,

It seems to be a bug from the arduino/esp8266 sdk. I reported a time ago. One user upload a fix for the version 2.3.0. Four days ago they released the 2.4.0 final version and after a quick test the problem was fixed.

Issues reported:

https://github.com/esp8266/Arduino/issues/3537
https://github.com/esp8266/Arduino/issues/3517

handgear, 2018-01-08:
Updated to 2.4.0 and problem solved.
(but still can't get disconnected info in loop. other issue for this.)

closed 2018-01-08 by handgear