odelot/aws-mqtt-websockets #34: messageArrived handler - how to get and parse the topic

genetipetz, 2018-02-02:
How do you get the topicName from the messageArrived handler out of the MessageData argument?

I want to be able to know the particular topic that invoked the messageArrived callback.

NOTE: My subscribe function is loading several subscriptions during set-up so that I can get multiple topics calling the same messageArrived callback handler:

char *subscribeTopic[5] = {
  "$aws/things/" THING_NAME "/shadow/update/accepted",
  "$aws/things/" THING_NAME "/shadow/update/rejected",
  "$aws/things/" THING_NAME "/shadow/update/delta",
  "$aws/things/" THING_NAME "/shadow/get/accepted",
  "$aws/things/" THING_NAME "/shadow/get/rejected"
};

//subscribe to a mqtt topic(s)
void subscribe () {

      for (int i=0; i<5; i++) {
        int rc = client->subscribe(subscribeTopic[i], MQTT::QOS0, messageArrived);
        if (rc != 0) {
          Serial.print("rc from MQTT subscribe is ");
          Serial.println(rc);
          break;
        }
        Serial.println("MQTT Topic subscribed");
      }
    
}

When I try to see the topicName inside the messageArrived handler, I am getting NULL values:
  
For example, this just prints a blank line as if the topicName wasn't passed back in the MessageData argument.
      Serial.println(md.topicName.cstring);

virgilioaray, 2018-02-15:
Hello, I will try to help you, English is not my native language, so I hope that Google translate does its job well.

After more than 24 hours of testing with the library (aws-mqtt-websockets), after watching videos on youtube, reading the source code, making changes, then compiling and uploading the firmware about 30 times I think I have an idea that Maybe (just maybe) it can help you.

My wish was to be able to control a device from AWS IOT, and that the communication is bidirectional between the device and the cloud, that is to say that the device could read instructions and in turn respond to the cloud, the example program only presents the scenario that the device

I share my version of the example program, this reads the status of the object and in turn changes, the key is to create two subscriptions, one for "$ aws / things / nameofthings / shadow / get / accepted" and another one for "$ aws / things / nameofthings / shadow / update ", to the constant " aws_topic " I made it variable :)

You only have to call once every "subscription" for each topic, so I noticed the reply message comes to you after each invocation to the "publish" function in the messageArrived () function

[aws-mqtt-websocket-example2.ino.zip](https://github.com/odelot/aws-mqtt-websockets/files/1726246/aws-mqtt-websocket-example2.ino.zip)

Well this program changes a switch from on to off and reads the status, I have fun seeing the state change in the aws iot console, I hope you can review and understand it and help you with your problem.

odelot, 2018-04-15:
@genetipetz I've uploaded an example that uses pubsubclient mqtt client. They use another approach for subscribing topics and I think it is more reliable than paho.