odelot/aws-mqtt-websockets #35: "section type conflict" compile error

github-seanli, 2018-04-27:
I tried to compile aws-mqtt-websocket-example-pubsubclient and got the error below - "error: mem_debug_file causes a section type conflict with __c"
The issue should be the same as - https://github.com/esp8266/Arduino/issues/2078

I'd like to get your input on
1. from your readme, can you share how you successfully compiled the binary?
2. is the issue known to you? any thought fixing it?
Otherwise, I will try to workaround it instead

C:\Users\xxx\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\xxx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/include" "-IC:\Users\xxx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/lwip2/include" "-IC:\Users\xxx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\xxx\AppData\Local\Temp\arduino_build_788939/core" -c  -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DDEBUG_ESP_PORT=Serial -DDEBUG_ESP_CORE -DDEBUG_ESP_WIFI -DDEBUG_ESP_HTTP_UPDATE -DDEBUG_ESP_UPDATER -DDEBUG_ESP_OTA -DDEBUG_ESP_OOM -include "umm_malloc/umm_malloc_cfg.h" -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"   -DESP8266 "-IC:\Users\xxx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266" "-IC:\Users\xxx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\variants\nodemcu" "-IC:\Users\xxx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\libraries\ESP8266WiFi\src" "-IC:\Projects\esp8266\libraries\aws-sdk-arduino-master" "-IC:\Users\xxx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\libraries\Hash\src" "-IC:\Projects\esp8266\libraries\arduinoWebSockets-master\src" "-IC:\Projects\esp8266\libraries\pubsubclient-master\src" "-IC:\Projects\esp8266\libraries\aws-mqtt-websockets-master" "C:\Projects\esp8266\libraries\pubsubclient-master\src\PubSubClient.cpp" -o "C:\Users\xxx\AppData\Local\Temp\arduino_build_788939\libraries\pubsubclient-master\PubSubClient.cpp.o"
Compiling library "aws-mqtt-websockets-master"
"C:\Users\xxx\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\xxx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/include" "-IC:\Users\xxx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/lwip2/include" "-IC:\Users\xxx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\xxx\AppData\Local\Temp\arduino_build_788939/core" -c  -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DDEBUG_ESP_PORT=Serial -DDEBUG_ESP_CORE -DDEBUG_ESP_WIFI -DDEBUG_ESP_HTTP_UPDATE -DDEBUG_ESP_UPDATER -DDEBUG_ESP_OTA -DDEBUG_ESP_OOM -include "umm_malloc/umm_malloc_cfg.h" -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"   -DESP8266 "-IC:\Users\xxx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266" "-IC:\Users\xxx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\variants\nodemcu" "-IC:\Users\xxx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\libraries\ESP8266WiFi\src" "-IC:\Projects\esp8266\libraries\aws-sdk-arduino-master" "-IC:\Users\xxx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\libraries\Hash\src" "-IC:\Projects\esp8266\libraries\arduinoWebSockets-master\src" "-IC:\Projects\esp8266\libraries\pubsubclient-master\src" "-IC:\Projects\esp8266\libraries\aws-mqtt-websockets-master" "C:\Projects\esp8266\libraries\aws-mqtt-websockets-master\AWSWebSocketClient.cpp" -o "C:\Users\xxx\AppData\Local\Temp\arduino_build_788939\libraries\aws-mqtt-websockets-master\AWSWebSocketClient.cpp.o"
In file included from C:\Users\xxx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/Arduino.h:304:0,
                 from C:\Users\xxx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\libraries\ESP8266WiFi\src/WiFiClient.h:25,
                 from C:\Users\xxx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\libraries\ESP8266WiFi\src/ESP8266WiFi.h:39,
                 from C:\Projects\esp8266\libraries\arduinoWebSockets-master\src/WebSockets.h:133,
                 from C:\Projects\esp8266\libraries\arduinoWebSockets-master\src/WebSocketsClient.h:28,
                 from C:\Projects\esp8266\libraries\aws-mqtt-websockets-master\AWSWebSocketClient.h:7,
                 from C:\Projects\esp8266\libraries\aws-mqtt-websockets-master\AWSWebSocketClient.cpp:1:
C:\Users\xxx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/umm_malloc/umm_malloc_cfg.h:188:40: error: mem_debug_file causes a section type conflict with __c
 #define malloc(s) ({ static const char mem_debug_file[] ICACHE_RODATA_ATTR STORE_ATTR = __FILE__; malloc_loc(s, mem_debug_file, __LINE__); })
                                        ^
C:\Projects\esp8266\libraries\aws-mqtt-websockets-master\CircularByteBuffer.h:42:18: note: in expansion of macro 'malloc'
   data = (byte*) malloc (capacity);
                  ^
In file included from C:\Users\xxx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/Arduino.h:258:0,
                 from C:\Projects\esp8266\libraries\aws-mqtt-websockets-master\AWSWebSocketClient.h:4,
                 from C:\Projects\esp8266\libraries\aws-mqtt-websockets-master\AWSWebSocketClient.cpp:1:
C:\Users\xxx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/pgmspace.h:16:51: note: '__c' was declared here
 #define PSTR(s) (__extension__({static const char __c[] PROGMEM = (s); &__c[0];}))
                                                   ^
C:\Users\xxx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/WString.h:38:76: note: in definition of macro 'FPSTR'
 #define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))
                                                                            ^
C:\Users\xxx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/WString.h:39:34: note: in expansion of macro 'PSTR'
 #define F(string_literal) (FPSTR(PSTR(string_literal)))
                                  ^
C:\Projects\esp8266\libraries\aws-mqtt-websockets-master\AWSWebSocketClient.cpp:433:27: note: in expansion of macro 'F'
     begin (host,port,path,F("mqtt"));
                           ^
Using library ESP8266WiFi at version 1.0 in folder: C:\Users\xxx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\libraries\ESP8266WiFi 
Using library aws-sdk-arduino-master in folder: C:\Projects\esp8266\libraries\aws-sdk-arduino-master (legacy)
Using library Hash at version 1.0 in folder: C:\Users\xxx\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\libraries\Hash 
Using library arduinoWebSockets-master at version 2.1.0 in folder: C:\Projects\esp8266\libraries\arduinoWebSockets-master 
Using library pubsubclient-master at version 2.6 in folder: C:\Projects\esp8266\libraries\pubsubclient-master 
Using library aws-mqtt-websockets-master in folder: C:\Projects\esp8266\libraries\aws-mqtt-websockets-master (legacy)
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

odelot, 2018-05-05:
Hi @github-seanli . I took my wife's notebook with a fresh Arduino install, got the library and its 3 dependencies ([1](https://github.com/odelot/aws-sdk-arduino), [2](https://github.com/Links2004/arduinoWebSockets) and [3](https://github.com/knolleary/pubsubclient)) and everything compiled without error. 

Sorry about the delayed answer, but I just got free time today

github-seanli, 2018-05-07:
Thanks, may I know what your ESP module is?

My board is https://www.amazon.com/HiLetgo-Internet-Development-Wireless-Micropython/dp/B010O1G1ES

I just tried it from scratch on my newly installed Ubuntu 16.04
1. I follow the steps mentioned in https://www.amazon.com/HiLetgo-Internet-Development-Wireless-Micropython/dp/B010O1G1ES to set up connection with ESP board ---no issue
Arduino IDE version is 1.8.5
2. Follow steps mentioned in https://www.arduino.cc/en/Guide/Libraries -  "Importing a .zip Library", download zip files from 3 dependencies (1, 2 and 3) as you pointed and then download zip from https://github.com/odelot/aws-mqtt-websockets and import into Arduino
3. Go to "File" -> "Examples" -> "aws-mqtt-websockets-master" -> "aws-mqtt-websockets-example-pubsubclient". ------ same compile error

My first post was collected from Windows 10 system.