odelot/aws-mqtt-websockets #37: Esp8266 - Long time to establish connection to AWS and messages each second.

junym, 2018-05-24:
Hi guys, firstly, congratulation to  for this excelent work of integration with aws iot.  
I'm not a expert about this, I'm woking with esp8266-01 and esp8266-12E.
My first issue is related to the initial connection with aws. Both my ESPs take about 5 minutes to  connect  with aws in port 443. After this, it's great the sending and receiving messasges. My doubt is related to this time of connection, why this happens?  This time is closed or can I change it?  After a hardware reset,  some times esp8266 take some seconds to reconnect, but sometimes it takes about 4/5 minutes like before.  

My second issue is when my esp is connected. Setting the parameter "debug Level" in my IDE arduino to core, I  notice my ESP sends/receive to AWS each 15 seconds. I don't understanding well this.  I'm using lib pubsubclient for mqtt  (https://github.com/knolleary/pubsubclient.git). I saw in the code "PubSubClient.cpp" at function  "boolean PubSubClient::loop()" that the author use  this defines: MQTT_MAX_PACKET_SIZE and 
MQTT_KEEPALIVE, but I'm not sure if just changing it solve my problem.  Can I change this parameter as I want?

>> The message I receive while ESP try connecting to aws:

[WS-Client] connection to a33yv9okseqbmj.iot.us-east-1.amazonaws.com:443 Faild
[WS-Client] client disconnected.
[WS-Client] client disconnected.
[WS-Client] connect wss... 
[WS-Client] connected to a33yv9okseqbmj.iot.us-east-1.amazonaws.com:443.
[WS-Client][sendHeader] sending header...
[WS-Client][sendHeader] handshake GET /mqtt?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIZKJYVJSYIF7DOBQ%2F19700101%2Fus-east-1%2Fiotdevicegateway%2Faws4_request&X-Amz-Date=19700101T080059Z&X-Amz-Expires=86400&X-Amz-SignedHeaders=host&X-Amz-Signature=95f0b4d95b125fc6c16c0443c8f263af67a8f3fefb59d1366f0cb00643efd8ab HTTP/1.1
Host: a33yv9okseqbmj.iot.us-east-1.amazonaws.com:443
Connection: Upgrade
Upgrade: websocket
Sec-WebSocket-Version: 13
Sec-WebSocket-Key: yuPjAE9716yxqFeG0pUDdw==
Sec-WebSocket-Protocol: mqtt
Origin: file://
User-Agent: arduino-WebSocket-Client
[WS-Client][sendHeader] sending header... Done (701616us).
[WS-Client][handleHeader] RX: HTTP/1.1 403 Forbidden
[WS-Client][handleHeader] RX: content-type: application/json
[WS-Client][handleHeader] RX: content-length: 163
[WS-Client][handleHeader] RX: date: Thu, 24 May 2018 18:11:31 GMT
[WS-Client][handleHeader] RX: x-amzn-RequestId: c6309e93-8b63-6bf9-7c2d-45c5d97843ab
[WS-Client][handleHeader] RX: connection: keep-alive
[WS-Client][handleHeader] RX: x-amzn-ErrorType: ForbiddenException:
[WS-Client][handleHeader] RX: access-control-allow-origin: *
[WS-Client][handleHeader] RX: access-control-expose-headers: x-amzn-ErrorMessage
[WS-Client][handleHeader] RX: access-control-expose-headers: x-amzn-RequestId
[WS-Client][handleHeader] RX: access-control-expose-headers: x-amzn-ErrorType
[WS-Client][handleHeader] RX: access-control-expose-headers: Date
[WS-Client][handleHeader] Header read fin.
[WS-Client][handleHeader] Client settings:
[WS-Client][handleHeader]  - cURL: /mqtt?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIZKJYVJSYIF7DOBQ%2F19700101%2Fus-east-1%2Fiotdevicegateway%2Faws4_request&X-Amz-Date=19700101T080059Z&X-Amz-Expires=86400&X-Amz-SignedHeaders=host&X-Amz-Signature=95f0b4d95b125fc6c16c0443c8f263af67a8f3fefb59d1366f0cb00643efd8ab
[WS-Client][handleHeader]  - cKey: yuPjAE9716yxqFeG0pUDdw==
[WS-Client][handleHeader] Server header:
[WS-Client][handleHeader]  - cCode: 403
[WS-Client][handleHeader]  - cIsUpgrade: 0
[WS-Client][handleHeader]  - cIsWebsocket: 0
[WS-Client][handleHeader]  - cAccept: 
[WS-Client][handleHeader]  - cProtocol: mqtt
[WS-Client][handleHeader]  - cExtensions: 
[WS-Client][handleHeader]  - cVersion: 0
[WS-Client][handleHeader]  - cSessionId: 
[WS-Client][handleHeader] no Websocket connection close.

>> The message ESP receive each 15 secondes while it's connected  : 

[WS][0][sendFrame] ------- send message frame -------
[WS][0][sendFrame] fin: 1 opCode: 2 mask: 1 length: 2 headerToPayload: 0
[WS][0][sendFrame] pack to one TCP package...
[WS][0][sendFrame] sending Frame Done (48676us).
[WS][0][handleWebsocketWaitFor] size: 2 cWsRXsize: 0
[readCb] n: zu t: 2
[WS][0][handleWebsocketWaitFor][readCb] size: 2 ok: 1
[WS][0][handleWebsocket] ------- read massage frame -------
[WS][0][handleWebsocket] fin: 1 rsv1: 0 rsv2: 0 rsv3 0  opCode: 2
[WS][0][handleWebsocket] mask: 0 payloadLen: 2

odelot, 2018-05-24:
Hi! I didn't have this 5 minutes to connect issue. I will try to observe my example to see if this happen. 

about the message every 15 seconds, this is the ping message from the mqtt protocol to keep the connection alive. you can configure it, as you did directly in the pubsubclient file. 

about the AWS keepalive configuration (search keep-alive-interval) [https://docs.aws.amazon.com/general/latest/gr/aws_service_limits.html](https://docs.aws.amazon.com/general/latest/gr/aws_service_limits.html)

> For MQTT (or MQTT over WebSockets) connections, a client can request a keep-alive interval between 30 - 1200 seconds as part of the MQTT CONNECT message. AWS IoT starts the keep-alive timer for a client when sending CONNACK in response to the CONNECT message. This timer is reset whenever AWS IoT receives a PUBLISH, SUBSCRIBE, PING, or PUBACK message from the client. AWS IoT will disconnect a client whose keep-alive timer has reached 1.5x the specified keep-alive interval (i.e., by a factor of 1.5).The default keep-alive interval is 1200 seconds. If a client requests a keep-alive interval of zero, the default keep-alive interval will be used. If a client requests a keep-alive interval greater than 1200 seconds, the default keep-alive interval will be used. If a client requests a keep-alive interval shorter than 30 seconds but greater than zero, the server treats the client as though it requested a keep-alive interval of 30 seconds.

junym, 2018-06-09:
Thanks  @odelot  for your response! Sorry for retard in my feedback!  Your link maked the keep alive procedure  very clear for me!  My esp get 3 min for connection establish for begin, but after some resets  it's taking some seconds to connecting.