odelot/aws-mqtt-websockets #40: Failed to connect with AWS

trungdn, 2018-06-22:
Hi, I double check the endpoint, port, key, secret. 
Grant IoTFull control for users. 

I received this error which looks like it could not connect to the Server.
pm open,type:0 0
failed, rc=-280152 - conn: 2 - (33272)

Please help what else could I check to know what is my problems.
Thank you very much.

debsahu, 2018-07-01:
@odelot same issue with me. any advice on how to solve the connection issue?

lunanigra, 2018-09-05:
Hi, I'm also running on this issue... Don't know what is going wrong exactly.

```
WiFi connected
IP address: 
172.23.56.222
8664 - conn: 1 - (44544)
pm open,type:0 0
failed, rc=-2
59671 - conn: 2 - (42704)
failed, rc=-2
110680 - conn: 3 - (42672)
```

Any help/tips welcome!

chan71, 2019-06-26:
Same issue here.
`21:21:24.319 -> rd l⸮⸮| ⸮l⸮|⸮l⸮b|⸮⸮⸮⸮⸮|"⸮p⸮p⸮n⸮loN⸮⸮⸮cp⸮⸮#$`rlp⸮N⸮⸮l⸮⸮bN⸮|l⸮⸮"⸮⸮no⸮l⸮⸮l`⸮2Nn$`or⸮⸮⸮Nb⸮lr⸮⸮⸮l⸮⸮d$⸮⸮{l⸮⸮o⸮connecting to wifi
21:21:26.205 -> 
21:21:26.205 -> connected
21:21:27.183 -> 3067 - conn: 1 - (47768)
21:21:34.345 -> pm open,type:0 0
21:22:18.280 -> failed, rc=-255167 - conn: 2 - (18792)
21:23:11.754 -> failed, rc=-2107324 - conn: 3 - (18760)
21:24:12.220 -> failed, rc=-2158864 - conn: 4 - (18728)
21:24:55.763 -> failed, rc=-2211352 - conn: 5 - (18696)
21:25:47.229 -> failed, rc=-2262798 - conn: 6 - (18664)
21:26:39.361 -> failed, rc=-2314940 - conn: 7 - (18632)
21:27:32.961 -> failed, rc=-2365952 - conn: 8 - (45480)
21:28:22.042 -> failed, rc=-2417623 - conn: 9 - (18568)
`

davidOSUL, 2020-01-26:
in case it helps anyone else https://github.com/odelot/aws-mqtt-websockets/issues/41#issuecomment-578488151