odelot/aws-mqtt-websockets #44: Aws-mqtt-websockets device stopped working

virgilioaray, 2018-08-17:
Hello, something very strange happened to me, for months I was running an iot device based on ESP8266 that was connected (by aws-mqtt-websockets) to AWS to read instructions from the shadow of a thing, but since yesterday it does not work, I thought it was the hardware but everything is fine, checking the programming I realize that I can no longer establish the connection to the server websocket, I checked the credentials (IAM) and I see that they have 183 days of creation, so I tried new credentials but the problem was not solved. Any ideas or suggestions?

odelot, 2018-08-18:
strange.... my devices are working normally. have you reviewed the permission in the user's policy?

WeaVaer, 2018-09-02:
Exactly the same thing happened to me. My code was working by the end May 2018 and I left it there.
After 3 months without doing any modification to my sketch or lib setup whatsoever, I gave it another go but got all the fancy error messages mentioned here to my frustration.
After fishing around for a remedy and trying many things that has been suggested here, finally the solution appeared in my case.
I had upgraded all the dependency libs in this repo but that didn't work.
Finally after upgrading my Arduino ESP8266 core lib to 2.4.2 from 2.3.0, everything started to work now.
I think both the lib and ESP8266 core upgrades did the job.
FYI

nzchicken, 2019-01-08:
I was getting a similar problem, and it looks like AWS might have changed their message that was being sent on update/accepted. PubSubClient has a limit of 128 characters by default, and it looks like it doesn't like it when a message is longer than that. Putting `#define MQTT_MAX_PACKET_SIZE 512` at the top before the `#include <PubSubClient.h>` should make life a bit easier.

The default message from this repo causes a response of 193 characters from AWS.