odelot/aws-mqtt-websockets #49: pubsubclient sample cannot compile cannot declare variable 'awsWSclient' to be of abstract type 'AWSWebSocketClient'

gunnicus, 2018-12-21:
aws-mqtt-websocket-example-pubsubclient:44: error: cannot declare variable 'awsWSclient' to be of abstract type 'AWSWebSocketClient'

 AWSWebSocketClient awsWSclient(1000);

                    ^

In file included from D:\Arduino\Wifi\aws-mqtt-websockets-master\examples\aws-mqtt-websocket-example-pubsubclient\aws-mqtt-websocket-example-pubsubclient.ino:21:0:

sketch\AWSWebSocketClient.h:25:7: note:   because the following virtual functions are pure within 'AWSWebSocketClient':

 class AWSWebSocketClient : public Client, private WebSocketsClient {

       ^

In file included from C:\Users\water\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta1\libraries\ESP8266WiFi\src/WiFiClient.h:27:0,

                 from C:\Users\water\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta1\libraries\ESP8266WiFi\src/ESP8266WiFi.h:39,

                 from D:\Arduino\Wifi\aws-mqtt-websockets-master\examples\aws-mqtt-websocket-example-pubsubclient\aws-mqtt-websocket-example-pubsubclient.ino:4:

C:\Users\water\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta1\cores\esp8266/Client.h:29:21: note: 	virtual int Client::connect(const IPAddress&, uint16_t)

         virtual int connect(CONST IPAddress& ip, uint16_t port) =0;

                     ^

C:\Users\water\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta1\cores\esp8266/Client.h:37:22: note: 	virtual bool Client::flush(unsigned int)

         virtual bool flush(unsigned int maxWaitMs = 0) = 0;

                      ^

C:\Users\water\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta1\cores\esp8266/Client.h:38:22: note: 	virtual bool Client::stop(unsigned int)

         virtual bool stop(unsigned int maxWaitMs = 0) = 0;

                      ^

exit status 1
cannot declare variable 'awsWSclient' to be of abstract type 'AWSWebSocketClient'

nzchicken, 2018-12-23:
I have just run into the same issue, however I believe this is due to the changes in the ESP8266 library. If you did what I did (cloned from their live github version) then you'll find the code isn't compatible for this library anymore. I would have thought that this would be a major version for them since they're changing the function declarations, but oh well.

My suggestion would be to checkout the 2.4.2 version of the ESP8266 repo, as that appears to compile alright with this library as it currently stands. Note, I needed to restart the arduino IDE in order to get this to verify

BMAnush, 2019-01-09:
I am facing the same issue.....What are the workarounds until the fix or next release?

stavros-zavrakas, 2019-02-14:
@nzchicken that worked for me

apicquot, 2019-02-19:
latest version of the esp8266/arduino library is using different signatures for the flush and stop methods: just need to change the signature of these 2 methods. I will make a pull request with my code. Yuo can otherwise directly download the aws-mqtt-websockets from my fork.

odelot, 2019-02-20:
thank you @apicquot. Have you tested with esp8266 sdk version 2.5.0 beta? and about websockets library and pubsubclient library version? everything is ok? I won't have time to test it. But i will update the readme with your info ^.^

apicquot, 2019-02-20:
I ll confirm to you the libraries version I am using tonight.

I posted an issue about erratic connection. Let me know if you have any clue about this.

Kind regards.

odelot, 2019-02-21:
@apicquot I've tested today and I had strange behavior with 2.5.0 sdk. When I turned on the debug messages from websocket library I start to got memory exceptions and reboots. Maybe the changes between 2.4.2 and 2.5.0 are big and will demand a close study to make the library works with it.

then I come back to 2.4.2 and my project was back to work fine. 

I will try to update the pubsubclient to version 2.7 and try the 2.5.0 sdk again later.

apicquot, 2019-02-21:
Thank you Fabio. I did not a chance to do test yesterday. I ll let you know if I find anything. Not easy at this point to know what issue is related to 2.5 , I am using a staged version between 2.4.2 and 2.5

Best.

apicquot, 2019-02-22:
I managed to get it work by changing the time source to
Amazon _useAmazonTimestamp = true, thank you for tip.

I am using
PubSubClient 2.6
AWS_SDK_Arduino master
Arduino Websocket 2.1.1
esp8266/Arduino https://github.com/apicquot/Arduino.git
tag=WebSocket_Working

Unfortunatly a bunch of error comes with current the current stage,
probably a bunch more of pure virtual method to adapt.

Best.

saamirt, 2019-03-14:
Can someone link to the esp8266 library 2.4.2 version that helped solve some issues? I'm not sure which library the previous comments were referring to.

apicquot, 2019-03-14:
You can pull it from my fork: https://github.com/apicquot/Arduino.git

from there you need to check out the branch Alex-Prod