odelot/aws-mqtt-websockets #52: Putting a private key on the device represents a HUGE security flaw

reidkersey, 2019-01-24:
The ESP8266 is capable of using TLS 1.2 via the use of WiFiClientSecure, and then adding in IoT certificates properly via the use of espClient.setCertificate, espClient.setPrivateKey & espClient.setCACert.

While this project still works for people using older versions of the ESP8266, the readme should be amended to state that a proper connection using TLS 1.2 can now be accomplished. Examples of how this is done can be found here: https://github.com/copercini/esp8266-aws_iot

odelot, 2019-01-28:
I updated the readme!

You don't need to use your private key, you can use AWS STS Temporary Credential, obtained from a web-service that uses the private key to generate temporary credentials.

apicquot, 2019-02-20:
Thew TLS1.2 with certificate works but is super slow (~7seconds). Your solution may still be a very good option.

WeaVaer, 2019-04-21:
@odelot 
I am a bit confused here,

1) Does that mean copercini's solution replaces the use of your aws-mqtt-websockets library and its dependencies which I was using succesfully for some time ?

2) Even if that works, using TLS1.2 means upgrading the ESP8266 core and replace your proven stable version with a one that may still have un-ironed issues,  with the added fact that you have to create 3 certs per each new device (a Thing on AWS) and run a new device provision algorithm somewhere.; i.e. added complexity. So I'd prefer to work with AWS STS temporary certs and a web service solution; much easier to pull and also I would not have to store private keys on my device. Is my thinkng logical and sound ? or I'm missing something. Please elaborate

3) Can u provide a basic example of your proposal about using AWS STS temp. credentials obtained from a webservice ? Although I'm familiar with doing exactly that on a web app using javascript, paho mqtt library, running a web service to AWS STS to provide temp. certficates etc, but I'm lost on how to implement it a) with your library? b) any other code sample?

Thanks a lot in advance

WeaVaer, 2019-05-03:
It seems that we will have to upgrade ESP8266 core to 2.5.0 in order to be able to continue working with AWS and need to do it fast since the last change to mqtt services that was rolled-back maybe rolled forward real soon.  I hope ESP core 2.5.0 has become mature and bug-free enough to do the migration.