odelot/aws-mqtt-websockets #61: How to connect by certificates instead of aws_key and aws_secret

miguelcsouza, 2020-09-19:
How could I connect to AWS using the certificates in this AWSWebSocketClient library instead of using aws_key and aws_secret?

odelot, 2020-09-21:
it seems that it is not possible to connect by websocket using x509 certificate (more [here](https://docs.aws.amazon.com/iot/latest/developerguide/protocols.html), tou can use a custom authentication or Signature Version 4 authentication (the one that this library uses, because it is TLS 1.1)

if you don't want to embbed a permanet aws key and secret, you can build a API to generate temporary aws keys using AWS STS. This library supports temporary aws keys made by TLS.

closed 2020-09-27 by miguelcsouza

WeaVaer, 2020-11-20:
@odelot 

The pub-sub example in version 1.3 of the library does not work for me.
The supplied root ca was being rejected.
But changing the root ca and ..
wham !
it works..
I can supply the working root ca to you if you want to update.

odelot, 2020-11-20:
@WeaVaer  thank you for the feedback... you can supply me with the root ca or you can submit a PR.

WeaVaer, 2020-11-20:
My region is 'us-east-1'
and I'm using your library on both an ESP8266 but also on an ESP32 :)

I have decoded the ca in the pub-sub example as an AWS Root CA 1 but that cert. and also other valid AWS Root CA 1 certs that I tried got rejected by AWS. The 'here' that you had commented previously tells it all about it. Thank you.

Then I found this root ca cert. in another working example of AWS mqtt connections and applied it to our example and.. bingo. 
It works. I decoded that cert as a "VeriSign Class 3 G5" valid to 2046.

Throwing in a PR..

odelot, 2020-11-21:
thank you @WeaVaer ^.^