odelot/aws-mqtt-websockets #62: Exception 3: LoadStoreError: Processor internal physical address or data error during load or store

AndriiNaboka, 2020-09-21:
Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
PC: 0x4000bf64
EXCVADDR: 0x40256ca9

Decoding stack results
0x402074d5: WebSocketsClient::loop() at /home/andy/Arduino/libraries/arduinoWebSockets-master/src/WebSocketsClient.cpp line 174
0x4020228a: AWSWebSocketClient::connect(char const*, unsigned short) at /tmp/arduino_build_6534/sketch/AWSWebSocketClient.cpp line 410
0x4020a7c1: uart_write(uart_t*, char const*, size_t) at /home/andy/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/uart.cpp line 509
0x40100cad: check_poison_neighbors(uint16_t) at /home/andy/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/umm_malloc/umm_local.c line 71
0x4020803c: HardwareSerial::write(unsigned char const*, unsigned int) at /home/andy/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/HardwareSerial.h line 164
0x40100d69: umm_malloc_core(size_t) at /home/andy/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/umm_malloc/umm_malloc.cpp line 458
0x402075a1: Print::write(char const*) at /home/andy/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/Print.h line 62
0x40207970: PubSubClient::connect(char const*, char const*, char const*, char const*, unsigned char, bool, char const*, bool) at /home/andy/Arduino/libraries/PubSubClient/src/PubSubClient.cpp line 196
0x4010050a: pvPortMalloc(size_t, char const*, int) at /home/andy/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/heap.cpp line 267
0x40207b60: PubSubClient::connect(char const*) at /home/andy/Arduino/libraries/PubSubClient/src/PubSubClient.cpp line 167
0x402023a6: generateClientID() at /media/andy/804D-0E5C/arduino-esp-test/aws_test/aws_test/aws_test.ino line 83
0x40202474: connect() at /media/andy/804D-0E5C/arduino-esp-test/aws_test/aws_test/aws_test.ino line 120
0x402026a0: setup() at /media/andy/804D-0E5C/arduino-esp-test/aws_test/aws_test/aws_test.ino line 193
0x402096f8: loop_wrapper() at /home/andy/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 194

ESP8266 - core version 2.7.4.

miguelcsouza, 2020-09-21:
Hello @AndriiNaboka all well?
I had the same problem and solved it by updating the "ESP8266 Community" card to version 2.7.4
to update open Boards Manager from Tools > Board menu and install esp8266 platform

AndriiNaboka, 2020-09-21:
Hi, @miguelcsouza 
I have 2.7.4 core version installed. You can see it in the /esp8266/2.7.4/