odelot/aws-mqtt-websockets #66: Unable to compile

TryingDev06, 2022-01-16:
I get the following errors when I try to compile the code. I have installed all of the dependencies. I'm not sure what's going wrong.

`/Users/username/Documents/Arduino/libraries/aws-mqtt-websockets-master/AWSWebSocketClient.cpp: In member function 'char* AWSWebSocketClient::generateAWSPath(uint16_t)':
/Users/username/Documents/Arduino/libraries/aws-mqtt-websockets-master/AWSWebSocketClient.cpp:210:16: error: no match for 'operator+=' (operand types are 'String' and 'char [(<anonymous> + 1)]')
  210 |  key_credential+=credentialScope;
      |  ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~
In file included from /Users/username/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/Arduino.h:286,
                 from /Users/username/Documents/Arduino/libraries/aws-mqtt-websockets-master/AWSWebSocketClient.h:4,
                 from /Users/username/Documents/Arduino/libraries/aws-mqtt-websockets-master/AWSWebSocketClient.cpp:1:
/Users/username/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/WString.h:133:17: note: candidate: 'template<class T> String& String::operator+=(const T&)'
  133 |         String &operator +=(const T &rhs) {
      |                 ^~~~~~~~
/Users/username/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/WString.h:133:17: note:   template argument deduction/substitution failed:
/Users/username/Documents/Arduino/libraries/aws-mqtt-websockets-master/AWSWebSocketClient.cpp:210:18: note:   variable-sized array type 'char [(<anonymous> + 1)]' is not a valid template argument
  210 |  key_credential+=credentialScope;
      |                  ^~~~~~~~~~~~~~~
exit status 1`

Any help or advice would be appreciated!

odelot, 2022-01-17:
Hi, the lib was tested with esp/Arduino SDK version 2.7.3 and you are using 3.0.2. maybe this is the reason.

ncastrinos, 2022-02-25:
Downgrading in the board manager to 2.7.3 removed this error for me.

closed 2022-02-25 by odelot