odelot/aws-mqtt-websockets #68: Compiling error: no match for 'operator+=' (operand types are 'String' and 'char [(<anonymous> + 1)]')

mnour94, 2022-04-24:
C:\Users\m\Documents\Arduino\libraries\aws-mqtt-websockets-master\AWSWebSocketClient.cpp: In member function 'char* AWSWebSocketClient::generateAWSPath(uint16_t)':

C:\Users\m\Documents\Arduino\libraries\aws-mqtt-websockets-master\AWSWebSocketClient.cpp:210:16: error: no match for 'operator+=' (operand types are 'String' and 'char [(<anonymous> + 1)]')

  210 |  key_credential+=credentialScope;

      |  ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~

In file included from C:\Users\maria\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/Arduino.h:286,

                 from C:\Users\m\Documents\Arduino\libraries\aws-mqtt-websockets-master\AWSWebSocketClient.h:4,

                 from C:\Users\m\Documents\Arduino\libraries\aws-mqtt-websockets-master\AWSWebSocketClient.cpp:1:

C:\Users\ma\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/WString.h:133:17: note: candidate: 'template<class T> String& String::operator+=(const T&)'

  133 |         String &operator +=(const T &rhs) {

      |                 ^~~~~~~~

C:\Users\m\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/WString.h:133:17: note:   template argument deduction/substitution failed:

C:\Users\m\Documents\Arduino\libraries\aws-mqtt-websockets-master\AWSWebSocketClient.cpp:210:18: note:   variable-sized array type 'char [(<anonymous> + 1)]' is not a valid template argument

  210 |  key_credential+=credentialScope;

      |                  ^~~~~~~~~~~~~~~

exit status 1

Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

kanaksony, 2022-05-20:
I am also facing the same issue.
Arduino Version: 1.8.19 (Windows Store 1.8.57.0)
@mnour94 Do you find any solution?

JohanM84, 2022-06-23:
Same issue here. Any suggestions?